esphome/feature-requests #2107: Add more sensors to the new ld2140 component

regevbr, 2023-02-08:
**Describe the problem you have/What new integration you would like**
A couple of days ago, @sebcaps added a cool ld2410 integration. I noticed that the [Home Assistant BLE integration](https://www.home-assistant.io/integrations/ld2410_ble) exposes more information about the gates, which can help calibrate the component.

**Please describe your use case for this integration and alternatives you've tried:**
Currently, in order to easily calibrate we need to download the radar app and do it over Bluetooth (you need to disconnect the rx/tx pins for the Bluetooth to start working)

**Additional context**
None

sebcaps, 2023-02-08:
Hello, thanks for your input. The idea was already mentioned  ([here](https://github.com/esphome/esphome/pull/3919#issuecomment-1301251044).I guess this is the same topic.
I'm not convinced it would be more helpful than a try/error for configuration or using the tool : data will be transmitted at a high rate and in my opinion hardly usable in HA for fine tuning.

regevbr, 2023-02-08:
Thanks, @sebcaps for the prompt response :-)  Adding @TheOfficialMrBlah to the mix as well, as he also wanted that feature in the original PR.

I agree this is too much into on HA, but so so does most of the sensors you added in the integration - I just use the presence one. Since this extra info is only sent in engineering mode, we can add a button to activate/disdable it so that way it will only be used when necessary, without installing random untrusted apks and disconencting the module from the uart in order to adjust it.
To make the feature more realistic, we need a way to configure the thresholds dynamically as well (not sure it is possible via esphome) and then the configuration process will be:
1. enable engineering mode
2. monitor the gates values
3. change the gates thresholds and repeat step 2 if necessary (without reflashing esp)
4. disable engineering mode

What do you think?

sebcaps, 2023-02-13:
Hello, indeed I could give a try, and let user choose if they wish use it or not. For sure, a switch would be the best way.
If you feel comfortable don't hesitate to submit a PR 😉

regevbr, 2023-02-14:
Cool already on it - managed to complete half of the features + a lot of nice extra stuff

ssieb, 2023-10-03:
Fixed by https://github.com/esphome/esphome/pull/4434

closed 2023-10-03 by ssieb