esphome/feature-requests #2110: BSEC Library update for bme680 (BSEC library version 1.6.1480 to 1.7.1492)

twenta, 2023-02-09:
The Bosch proprietry library for bme680 was updated recently.
Esphome uses the old version ([1.6.1480](https://github.com/boschsensortec/BSEC-Arduino-library/releases/tag/v1.6.1480)). As this version is the first update within 3 years and fixes some bugs, I hope the accuracy of the sensor is improved.

https://www.arduinolibraries.info/libraries/bsec-software-library
https://github.com/boschsensortec/BSEC-Arduino-library/releases/tag/v1.7.1492

nagyrobi, 2023-02-09:
@trvrnrth @bisbastuner

kiwipaulrob, 2023-02-11:
plus 1

snitkjaer, 2024-04-11:
can we somehow help with this?

bisbastuner, 2024-05-05:
I started working on this on the past week.
The first thing that was clear is that we cannot use version 1.7.1492 because that's not available in PlatformIO ([see here](https://registry.platformio.org/libraries/boschsensortec/BSEC%20Software%20Library/versions)). Or, at least I don't know how to do it.
So I started integrating version 1.8.1492, which is also the latest release from BOSCH for this library (we now have the new BSEC2 library available).
I needed to change many parts of the code, since BOSCH has moved from using `bme680_*` functions to using `bme68x_*` functions (I think because of the added compatibility with BME688).
They also changed some units, in particular we don't need to divide temperature by 100 and RH by 1000 now (at least when using the floating point code path).
But I had some problems calling the `bsec_do_steps()` function, which is the heart of the BSEC library. It was constantly returning error -2.
Since that error is `BSEC_E_DOSTEPS_VALUELIMITS`, and the T,P,H values returned by `bme68x_get_data()` looked OK, I started experimenting with the gas_resistance value.
I discovered that `bsec_do_steps()` imposes a hard maximum limit for gas_resistance: 12800000 ohm is the maximum accepted value; anything above that will result in `bsec_do_steps()` failing with the `BSEC_E_DOSTEPS_VALUELIMITS` error.
My sensor seems to be returning a value of 12917167 with clean air, or at least right after turning it on. That's the value returned both when using the current (v1.6.1480) and the new (v1.8.1492) library from BOSCH.
It seems to me that the library has an internal bug, perhaps in the gas_resistance calculation, since I hope the new 12800000 limit has been introduced for sanity checking.

bisbastuner, 2024-05-05:
> can we somehow help with this?

I would benefit for some gas resistance readings from all of you, in order to exclude that my sensor is somewhat broken or misbehaving. It seems to work correctly with the current bme680_bsec integration, even with the gas resistance value of 12917167, which is not considered valid by the new library.
What are your readings?
I didn't log the gas_resistance value in Home Assistant, but will start logging mines in the next days to try and understand what's going on.

trvrnrth, 2024-05-10:
Here's the last month from three sensors in different rooms if that helps:
[image]

bisbastuner, 2024-05-12:
These are my readings in the last week
[image]
My readings are corrected with a 0.9909... correction factor, to circumvent the library value limit error.
They are quite high compared to yours, and they seem to be capped. I also note that the IAQ value output by the library can vary in response to RH and temperature, without a GasResistance change.
This is an outdoor sensor experimentally using the new bme680_bsec and the new BOSCH v1.8.1492 library; I'll enable GasResistance readings on my other sensor (indoor) and see what it does output (with the current bme680_bsec).

I can also commit my current work on GitHub if you want to review and/or test it.
I may need to add a temporary additional configuration option to insert the correction factor, so I can test it corrected; while you can test it without correction

bisbastuner, 2024-06-09:
These are the readings over the last month
[image]
espino2out (blue) is the outdoor sensor which is using the new experimental BOSC v1.8.1492 library, while espino1in (violet) is the indoor sensor, using the current official bme680_bsec component from ESPHome 2024.4.1

It seems that, after nearly one month, the GasResistance has been stabilizing to more "regular" values.
The indoor sensor shows values 1-2 order of magnitude lower than the outdoor one, but the outdoor sensor now shows values similar to your blue sensor @trvrnrth 
I think the sensor history comes in play here, as the library saves the state of the last 28 days.
I purposedly changed the ESPHome preference name, as I found out the BSEC_MAX_STATE_BLOB_SIZE has increased in the new library. I would expect ESPHome would not have found any preference, and so I should have started from scratch, but who knows...

By the way, I think my external sensor is not broken; we would need to closely investigate this behavior as to not lock-up users with this strange error.

I'll remove the correction factor in my code and start testing the library with untouched values, then I'll send a PR with a call for testing in the next days

bisbastuner, 2024-12-08:
These are my gas resistance readings over the last 6 months:
[image]
Again, espino2out (blue) is the outdoor sensor which is using the new experimental BOSC v1.8.1492 library, while espino1in (violet) is the indoor sensor, using the current official bme680_bsec component from ESPHome 2024.4.1

I have very very strange readings for the gas resistance: the blue one is hitting the saturation value at 12,8 Mohm, and the violet one seems to be too low, never reaching even 400 kohm.

This, combined with the fact that with the new library the component cannot be compiled anymore with ESP8266 even commenting out big parts of the code (it doesn't fit into the esp8266 RAM), and the fact that we now have the new [bme68x_bsec2 component](https://github.com/esphome/esphome/pull/4585) using the BOSCH library version 2, make me think that it's better to leave this component as-is, for backwards compatibility with the ESP8266 devices.
ESP32 owners, [and maybe even ESP8266 owners](https://community.home-assistant.io/t/full-guide-to-read-bme680-via-bsec2-on-esp8266/778505), can go ahead using the new component