esphome/feature-requests #2113: Expose Deep sleep wakeup cause

Lumpusz, 2023-02-11:
Esp32 can tell what was the cause of deep sleep wakeup. This is being used already internally to figure out different timeouts based on the cause, but it would be nice to expose this to an automation in on_boot.
Example use case: battery powered rain gauge, when it is woken up by GPIO because the bucket was tripping, capture rainfall data, when it is waken up by timeout, just measure temperature, wind speed, direction etc.

jul-fls, 2023-04-05:
Hi, is there any new informations about this feature progress ?
In my case i have an esp32 hooked up to 2 reed switches to sense my mailbox (1 reed switch for the mail slot and another reed switch for the package door) and it's continuously in deep sleep but when woken up the door or mail slot may have already been closed again so it doesn't detect why it has been woken up i use this configuration 
```yml
deep_sleep:
  id: deep_sleep_1
  run_duration: 30s
  sleep_duration: 86400s
  wakeup_pin_mode: INVERT_WAKEUP
  esp32_ext1_wakeup: 
    mode: ANY_HIGH
    pins: 
      - GPIO0
      - GPIO4
```
and i'm looking for a solution to tell to home assistant which pin has been triggered so i can detect if it's because of the mail slot or the package door, does someone have an idea how to do this ?
Thank you very much for looking into this.
Best regards.

meanmrgreen, 2023-05-24:
> 

I have this same problem, and a mailbox that needs this :)

I just tested a possible workaround.

If i create a binary template sensor 
```- platform: template
name: "${device_name} Pin wakeup"
id: pinwakeup
````
    
Then create a script that runs on boot that checks the state of my reed sensor as soon as the sensors are setup but before wifi and other stuff is setup.
````
  on_boot:
    priority: 600
    then: 
      - script.execute: pinwakeup
````

   ````
  - id: pinwakeup
    then:
    - if:
        condition:
         - binary_sensor.is_on: reedswitch
        then:
         - binary_sensor.template.publish:
             id: pinwakeup
             state: ON
        else:
         - binary_sensor.template.publish:
            id: pinwakeup
            state: OFF
  ````
            
This seems to be working pretty good, unless im really really fast opening and closing the reed switch, im talking maybe 0.1 seconds or something.  Then when the esp wakes up, connects to to homeassistant it will then send the on state from the reed sensor that woke it up.

Worked great with any opening that was around like half a second/second. Not perfect but still better than needing it to be open for 10-20s to register the state.

heyitsyang, 2023-12-22:
I have the same requirement as the OP.  I'd like to press a GPIO button that wakes the sleeping device for a longer wake time (and possibly run other code) than when the timer based wakeup is reached.  As the OP said, wakeup_reason is available in the native ESP code.

ssieb, 2023-12-28:
It's easy to get it using `rtc_get_wakeup_cause()`.  You can find the list of values at https://github.com/esphome/esphome/blob/dev/esphome/components/debug/debug_component.cpp#L277.  You will probably need to add the `debug:` component to bring in the necessary headers.