esphome/feature-requests #2114: Repurpose ESPHome devices in Home Assistant

anoblet, 2023-02-12:
**Describe the problem you have/What new integration you would like**
The ability to repurpose ESPHome devices within Home Assistant so that they retain their data.

**Please describe your use case for this integration and alternatives you've tried:**
I have a few Athom smart plugs that are able to monitor energy consumption. I often repurpose them, especially during summer and winter months for air conditioners and space heaters. If I upload a new configuration to a smart plug, it gets registered in Home Assistant as a new device under the same integration or name. In order to get energy reporting working for the new device, I have to remove the integration for the old device, losing historical data, and register the new device. It would be helpful if both devices could exist, even if one was considered offline thus retaining historical data.

**Additional context**
I'm not sure if this is an ESPHome issue or a Home Assistant issue.

nagyrobi, 2023-02-12:
When you repurpose, what do you change in the config?

anoblet, 2023-02-12:
Sample config:

```
substitutions:
  friendly_name: "Bedroom Heater"
  name: bedroom-heater
packages:
  athom.smart-plug-v2: github://athom-tech/athom-configs/athom-smart-plug-v2.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
->
```
substitutions:
  friendly_name: "Bedroom Air Conditioner"
  name: bedroom-air-conditioner
packages:
  athom.smart-plug-v2: github://athom-tech/athom-configs/athom-smart-plug-v2.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

nagyrobi, 2023-02-12:
To achieve the behavior you want, you should try to use different IP addresses for the device in their different scopes, so they become uniquely differentiated by HA.

Something like this added to your code above, and change `device_ip` too:

```yaml
substitutions:
  device_ip: 192.168.1.11

wifi:
  manual_ip:
    static_ip: ${device_ip}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  use_address: ${device_ip}
```

anoblet, 2023-02-12:
I use manual IP in the configuration to overwrite the first device with the
second device configuration.

In Home Assistant are ESPHome devices identified by IP or MAC?

I haven't spent time assigning static IP to my ESPHome devices. I will now.

Would it make more sense for Home Assistant to mDNS on IP, check the MAC,
and randomize MAC from ESPHome?

Thanks for the help :)

ezpl, 2023-02-14:
So, the uniqueness in HA is determined by IP-address? It seems strange that the esphome node name is not considered. I am trying to avoid using fixed IPs all over the place.
It would be great if this at least could be configurable, perhaps in an optional property in the 'esphome' section in yaml, like 'unique_id: my_esphome_device_01'

ssieb, 2023-02-14:
It's not the IP address.  It's at least the name, but maybe something else as well.

ssieb, 2023-02-14:
Are you saying that if you flash a config with a different name to the same device, it somehow still works under the old name?  That doesn't make sense.

anoblet, 2023-02-14:
@ssieb

When I upload a new configuration, there is no notification in Home Assistant for a new device. The devices gets added under the previous integration.

Every once and a while I will get a notification for a device that already exists, and when I check, there are two devices under the same integration.

nagyrobi, 2023-02-14:
Yeah indeed that happens exactly like that. If you change both name and IP address it's gonna be a different device.