esphome/feature-requests #2115: Please can the DFRobot 2-Channel I2C DAC module (DFR0971) be integrated into ESPHome

ledgardeneruk, 2023-02-12:
**Describe the problem you have/What new integration you would like**
The DFRobot 2-Channel I2C DAC module (DFR0971) be integrated into ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
It's very useful for people who have analogue 0-10v LED dimmer drivers that don't use PWM.

**Additional context**
Links to the Github from DFRobot with Arduino code:

- https://wiki.dfrobot.com/SKU_DFR0971_2_Channel_I2C_0_10V_DAC_Module#target_8
- https://github.com/DFRobot/DFRobot_GP8403

ledgardeneruk, 2023-02-15:
I have been looking at the code again and again it looks like there are only 3 variables mainly used: 

- setDACOutRange(eOutPutRange range);
- setDACOutVoltage(uint16_t data,uint8_t channel);

- 0-5v to 0-10v range
- set 0-10v and which channels.

ledgardeneruk, 2023-02-17:
I have also come across a more basic version of the .h file and its uses in IDE. I think it should be quite easy to do from this for someone in the know: https://github.com/porkyneal/GP8403

inspectorgrow, 2023-02-28:
I would also be interested in this.

AlexH269, 2023-03-26:
I'm using this device to control a ventilation system and would be interested in this getting added to ESPHome as well.

alxzndr, 2023-03-31:
+1 
I also want to use this module to control a ventilation system.

inspectorgrowtemp, 2023-03-31:
> +1 I also want to use this module to control a ventilation system.
> 
> Edit: Found this pull request: [esphome/esphome#4495](https://github.com/esphome/esphome/pull/4495)

Got a thread started here if you wanted to join: [Inspector Grow Forum](https://forums.inspectorgrow.com/t/gravity-2-channel-i2c-dac-module-0-10v-working-in-esphome-perfect-for-analog-0-10v-lighting-control/24)

closed 2023-05-08 by jesserockz