esphome/feature-requests #2116: Add Observability to slow_pwm - II - follow up of #998

maxnomo60, 2023-02-12:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

maxnomo60, 2023-02-12:
HI, following up issue #998 its written to get this info with a 

A) template ("You could use a template output in between to capture the value.")   - or -

B) with a "new slow_pwm which is not tied to output".  Both ideas i do not really understand what really to do.  

Pls specify how to do that ??

output:
  - platform: slow_pwm
    pin: GPIO02
    id: slow_pwm_out     # s.o. und  s.u. 
    period: 2s           # Periodendauer innerhalb der getaktet wird - bei FBH ca 5-8 min = 300-500 sec
#
    turn_on_action:
      output.turn_on: pidheat
    turn_off_action:
      output.turn_off: pidheat

and 

binary_sensor:
  - platform: template
    name: "pidheat"
    id: pidheat

ERROR MESSAGE AS: ID 'pidheat' of type template_::TemplateBinarySensor doesn't inherit from output::BinaryOutput. Please double check your ID is pointing to the correct value.

Any ideas how to solve  that ?

nagyrobi, 2023-02-12:
Please head on to our Discord support channel with such direct questions.

closed 2023-02-12 by nagyrobi