esphome/feature-requests #2120: Addressable CWWW support (based on WS2811)

steeve, 2023-02-15:
**Describe the problem you have/What new integration you would like**

CWWW light strips now come in addressable fashion, such as https://fr.aliexpress.com/item/1005005069644213.html. The two white channels (2700k and 6500k) are mapped are mapped to RED and GREEN channels.

**Please describe your use case for this integration and alternatives you've tried:**

Be able to set colour temperature for those addressable light strips.

**Additional context**

I think it's possible to emulate it via a custom light component, or light template in HA. But I'd wager it would make sense to add it directly in ESPHome, since it already has the `cwww` component. Perhaps the RED and GREEN channels could be exposed as output components?

steeve, 2023-02-15:
I sort of got it to work with the following custom light output, but I don't think it's write to create a `LightCall` at each `write_state`, since it seems to do networking back to HA, and causes the transition to be skipped completely, and spams the log.

Nevertheless, it demonstrates the point:
```yaml
light:
  - id: tmp1
    platform: neopixelbus
    variant: WS2811
    pin: GPIO18
    num_leds: 37
    name: "TMP LEDS"
    power_supply: leds_relay
    
  - platform: custom
    lambda: |-
        auto light_out = new AddressableCWWW(id(tmp1), 2700, 6500);
        App.register_component(light_out);
        return {light_out};
    
    lights:
      - name: "My Custom Light"
```

```cpp
#include "esphome.h"

class AddressableCWWW : public light::AddressableLight {
private:
    light::AddressableLight *src_;
    LightState *src_state_;
    int32_t cwt_;
    int32_t wwt_;

public:
    AddressableCWWW(light::LightState *src_state, int32_t cwt, int32_t wwt) :
        src_(static_cast<light::AddressableLight *>(src_state->get_output())),
        src_state_(src_state),
        cwt_(cwt),
        wwt_(wwt)
    {
    }

    void setup() override {
        this->src_->setup();
    }
    
    LightTraits get_traits() override {
        auto traits = LightTraits();
        traits.set_supported_color_modes({
            ColorMode::COLD_WARM_WHITE,
            ColorMode::BRIGHTNESS
        });
        traits.set_min_mireds(1000000 / this->wwt_);
        traits.set_max_mireds(1000000 / this->cwt_);
        return traits;
    }

    int32_t size() const {
        return this->src_->size();
    }

    void clear_effect_data() {
        this->src_->clear_effect_data();
    }

    void write_state(LightState *state) override {
        float cwt = 0.0f;
        float wwt = 0.0f;
        state->current_values_as_cwww(&cwt, &wwt, true);

        bool enabled = false;
        state->current_values_as_binary(&enabled);

        this->src_state_->make_call()
            .set_state(enabled)
            .set_blue(0.0f)
            .set_red(wwt)
            .set_green(cwt)
            .set_transition_length(0)
            .perform();
    }

protected:
    ESPColorView get_view_internal(int32_t index) const override {
        return this->src_->get(index);
    }
};
```

matthewmax09, 2023-12-06:
Hi there,

I have the same fcob cct strips as you, and tried to adapt your script for esp32_rmt_led_strip.

But ran into some errors and was wondering if you might have some tips on how I may tackle this?

```
error: expected type-specifier before 'AddressableCWWW'

error: could not convert '{light_out}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::light::LightOutput*>'
```

gartensofa, 2025-02-27:
Hi guys,
i would like to leave a snipped of my working-ish .yaml here,
for anyone's reference. No custom components required.

```
output:
  - platform: template
    id: output_cold
    type: float
    write_action:
      lambda: |-
        auto call = id(rgb).turn_on();
        call.set_green(state);
        call.perform();
  - platform: template
    id: output_warm
    type: float
    write_action:
      lambda: |-
        auto call = id(rgb).turn_on();
        call.set_red(state);
        call.perform();

light:
  - platform: cwww
    id: cw_ww
    name: "CWWW"
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3000 K
    default_transition_length: 0s
  - platform: neopixelbus
    type: GRB
    id: rgb
    variant: WS2811
    pin: GPIO4
    num_leds: 96
    name: "NeoPixel Light"
    default_transition_length: 1s
    internal: True
    method:
      type: esp32_rmt
```

Modify the method of the neopixelbus according to your esp-device.

I say working-ish, because with my led-strip it only starts to light up at around 50% set brightness and produces a dip in brightness when changing the light temperature from one extreme to the other.

pranelio, 2025-03-26:
I think i've managed to improve upon work by @gartensofa . The Gamma correction can be adjusted to your liking and the brightness works as it should.

Adjust the led count, pin and kelvin, depending on your setup.

`globals:
  - id: cw_calculated_value
    type: float
    restore_value: no
    initial_value: '0.0'
  - id: ww_calculated_value
    type: float
    restore_value: no
    initial_value: '0.0'
  - id: color_temperature_kelvin
    type: float
    restore_value: no
    initial_value: '2700'
  - id: brightness
    type: float
    restore_value: no
    initial_value: '0.0'

script:
  - id: update_led_strip
    mode: queued
    then:
      - lambda: |-
          if (!id(cw_ww).current_values.is_on()) {
            id(LED_Strip).turn_off().perform();
            return;
          }
          const float gamma = 0.5;
          id(color_temperature_kelvin) = id(cw_ww).current_values.get_color_temperature_kelvin();
          id(brightness) = id(cw_ww).current_values.get_brightness();
          id(cw_calculated_value) = (id(color_temperature_kelvin) - 2700.0) / (6500.0 - 2700.0);
          id(ww_calculated_value) = 1.0 - id(cw_calculated_value);
          id(cw_calculated_value) = pow(id(cw_calculated_value), gamma);
          id(ww_calculated_value) = pow(id(ww_calculated_value), gamma);
          id(cw_calculated_value) *= id(brightness);
          id(ww_calculated_value) *= id(brightness);

          auto call = id(LED_Strip).turn_on();
          call.set_green(id(cw_calculated_value));
          call.set_red(id(ww_calculated_value));
          call.perform();

output:
  - platform: template
    id: output_cold
    type: float
    write_action:
      lambda: |-
        id(update_led_strip).execute();

  - platform: template
    id: output_warm
    type: float
    write_action:
      lambda: |-
        id(update_led_strip).execute();

light:
  - platform: cwww
    id: cw_ww
    name: "Living Room LED"
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    default_transition_length: 0.5s

  - platform: neopixelbus
    type: GRB
    id: LED_Strip
    variant: WS2811
    pin: GPIO35
    num_leds: 100
    name: "NeoPixel Light"
    default_transition_length: 0s
    internal: true
    method:
      type: esp32_rmt
      channel: 0`

gartensofa, 2025-03-26:
After an initial test i can confirm, that works indeed better.
Thank you for sharing!

bryce-cahill, 2026-04-03:
@Buxxter or @pranelio This method was working great for a long time however, this morning I made the mistake of updating my device and it seems something has been broken by a new release, has anyone found a fix?

It seems that `get_color_temperature_kelvin()` is always returning 6500K and therefore the LED strip stays at cool white all the time. Replacing the call with `get_cold_white()` and `get_warm_white()` fixes the output of the LED strip but the state stays incorrect in Home Assistant, any ideas?

```
script:
  - id: update_led_strip
    mode: queued
    then:
      - lambda: |-
          if (!id(cw_ww).current_values.is_on()) {
            ESP_LOGD(TAG, "Light is OFF");
            id(LED_Strip).turn_off().perform();
            return;
          }

          id(cw_calculated_value) = id(cw_ww).current_values.get_cold_white();
          id(ww_calculated_value) = id(cw_ww).current_values.get_warm_white();
          id(brightness) = id(cw_ww).current_values.get_brightness();

          const float gamma = 0.5f;
          id(cw_calculated_value) = powf(id(cw_calculated_value), gamma);
          id(ww_calculated_value) = powf(id(ww_calculated_value), gamma);

          auto call = id(LED_Strip).turn_on();
          call.set_green(id(cw_calculated_value));
          call.set_red(id(ww_calculated_value));
          call.set_blue(0.0f);
          call.set_brightness(id(brightness));
          call.perform();
```

pranelio, 2026-04-07:
@bryce-cahill I do remember that some past update broken this implementation and some flickering appeared. My current implementation looks like this:

File `/homeassistant/esphome/common/cwww_to_ws2811.yaml`

```
substitutions:
  cwww_light_name: "CWWW Light"

  neopixel_pin: GPIO16
  neopixel_leds: "100"

  warm_temp: "2700"
  cold_temp: "6500"
  gamma: "0.5"

globals:
  - id: cw_calculated_value
    type: float
    restore_value: yes
    initial_value: '0.0'

  - id: ww_calculated_value
    type: float
    restore_value: yes
    initial_value: '0.0'

  - id: color_temperature_kelvin
    type: float
    restore_value: yes
    initial_value: ${warm_temp}

  - id: brightness
    type: float
    restore_value: no
    initial_value: '0.0'

script:
  - id: update_led_strip
    mode: queued
    then:
      - lambda: |-
          if (!id(cwww_light).current_values.is_on()) {
            id(neopixel_driver).turn_off().perform();
            return;
          }

          const float gamma = ${gamma};
          auto call = id(neopixel_driver).turn_on();

          id(color_temperature_kelvin) = id(cwww_light).current_values.get_color_temperature_kelvin();
          id(brightness) = id(cwww_light).current_values.get_brightness();

          id(cw_calculated_value) = (id(color_temperature_kelvin) - ${warm_temp}) / (${cold_temp} - ${warm_temp});
          id(ww_calculated_value) = 1.0 - id(cw_calculated_value);

          id(cw_calculated_value) = pow(id(cw_calculated_value), gamma);
          id(ww_calculated_value) = pow(id(ww_calculated_value), gamma);

          id(cw_calculated_value) *= id(brightness);
          id(ww_calculated_value) *= id(brightness);

          call.set_red(id(ww_calculated_value));
          call.set_green(id(cw_calculated_value));
          call.perform();

output:
  - platform: template
    id: output_cold
    type: float
    write_action:
      lambda: |-
        id(update_led_strip).execute();

  - platform: template
    id: output_warm
    type: float
    write_action:
      lambda: |-
        id(update_led_strip).execute();

light:
  - platform: cwww
    id: cwww_light
    name: ${cwww_light_name}
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: "${cold_temp}K"
    warm_white_color_temperature: "${warm_temp}K"
    default_transition_length: 0.5s

  - platform: neopixelbus
    type: GRB
    id: neopixel_driver
    variant: WS2811
    pin: ${neopixel_pin}
    num_leds: ${neopixel_leds}
    internal: true
    default_transition_length: 0s  
    # method:
    #   type: esp32_rmt
    #   channel: 0  

```

esphome device config:

```
esphome:
  name: bathroom-led
  friendly_name: Bathroom LED

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "YOUR ENCRYPTION KEY"

ota:
  - platform: esphome
    password: "YOUR OTA PASSWORD"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password   

packages:
  light_strip:
    !include
      file: common/cwww_to_ws2811.yaml
      vars:
        cwww_light_name: "Bathroom LED"
        neopixel_pin: GPIO35
        neopixel_leds: "21"
        warm_temp: "2700"
        cold_temp: "6500"
        gamma: "0.5"
```

This allows me to have multiple LED controllers running on the same core logic, but with different names and LED count.

And to enable the light, my automation action looks like this:

```
action: light.toggle
metadata: {}
data:
  color_temp_kelvin: 2710
  brightness_pct: 100
target:
  entity_id:
    - light.bathroom_led_bathroom_led
    - light.bathroom_mirror_bathroom_mirror_led
```