esphome/feature-requests #2121: Support for WTC401SPI touch sensor

lanrat, 2023-02-15:
**Describe the problem you have/What new integration you would like**

I'm requesting support for the WTC401SPI touch sensor. I've found this touch sensor in some ESP32 based LED lamps for the brightness and color controls.

[The datasheet and some example code can be found here.](http://en.wincomtech.com/Private/ProductFiles/5210978ec3799cfdd0d2.pdf)

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to replace the firmware on some of these LED lamps with ESPHome, but in order to get the inputs working, support for this chip would be required.

**Additional context**

One lamp that uses an ESP32 and the WTC401SPI is the dlamp.

lanrat, 2024-09-28:
Found a similar touch panel which may share similarities with the underlying library: https://github.com/racerxdl/wtc801

Tomcuzz, 2025-03-24:
So did some messing around with one of these and was actually able to get this working using a lambda in esphome
Code might need a bit of tidying up but a good starting point could be:
```
spi:
  - clk_pin: <SCK Pin>
    mosi_pin: <SDI Pin>
    miso_pin: <SDO Pin>
    interface: hardware

spi_device:
  - id: wtc401
    cs_pin: <SCS Pin>

binary_sensor:
  - platform: gpio
    name: "Touch Registered"
    pin:
      number: <Flag Out Pin>
      inverted: true
      mode:
        input: true
    filters:
      - delayed_on: 100ms
    on_press:
      then:
        - lambda: !lambda |-
            id(wtc401).enable();
            uint8_t data = id(wtc401).transfer_byte(0xFF);
            id(wtc401).disable();
            if (data == 131) {
              id(power_button).turn_on();
<There are other pins I don’t have hooked up but you can log to check these number>
            } else {
		uint8_t ndata = ((55 - data)* 255) / 53);
            }
    on_release:
      then:
        <Do something when finger released>
```

lanrat, 2025-03-24:
Hey @Tomcuzz, I haven't looked into this in a while. By any chance are you also looking at porting ESPHome to the dLight?

Tomcuzz, 2025-03-25:
Hey @lanrat sorry I not able to comment on that subject