esphome/feature-requests #2122: Opt-out from prefixing some entities with friendly name

felipecrs, 2023-02-16:
**Describe the problem you have/What new integration you would like**

I'm trying to migrate to the new Friendly Name introduced recently, but the main issue I found was:

I have a SONOFF 4CH, with 4 different lights wired in it.

I don't want each light prefixed with the friendly name of my device. Like:

Friendly Name: Sonoff Lights
Light 1: Bathroom Light
Light 2: Kitchen Light

Instead of:
Friendly Name: Sonoff Lights
Light 1: Sonoff Lights Bathroom Light
Light 2: Sonoff Lights Kitchen Light

**Please describe your use case for this integration and alternatives you've tried:**

The alternative is not to set a friendly name for the device.

**Additional context**

Weissnix4711, 2023-09-17:
Any update on this? Has anything changed re. friendly_name prefixing since Feb? I can't seem to figure out if this is possible now, so I'm gonna presume not. Would be nice not to have to change entity names individually within HA, but do it in firmware side instead.

Adrian57, 2024-04-08:
The friendly name prefixing thing is a total disaster. No one seems to understand the problem, why it occurs some times, others not, why sometimes the prefixing is applied, others not, why sometimes entitly names end up with the friendly prefix twice, sometimes not.
...and noone seems to want to do anything about it :-(

nagyrobi, 2024-04-08:
Very easy: don't use it. It's an optional parameter.

felipecrs, 2024-04-08:
> why sometimes entitly names end up with the friendly prefix twice

This just happened to my entities when I upgraded to latest 2024.3.2. Very odd indeed.

@nagyrobi thanks for your suggestion. The best way forward so far.

felipecrs, 2024-04-08:
> why sometimes entitly names end up with the friendly prefix twice

This just happened to my entities when I upgraded to latest 2024.3.2. Very odd indeed.

@nagyrobi thanks for your suggestion. For now, indeed best way forward is going backwards. 😅

felipecrs, 2024-04-08:
> This just happened to my entities when I upgraded to latest 2024.3.2.

PS: just for entities whose friendly names had special characters like `ã`, these special characters ended up being replaced with `_` instead of `a`, and therefore I got:

```
cover.portao_da_rua_port_o_da_rua
```

(in case someone wants to investigate it)

felipecrs, 2024-04-08:
😭 Thanks for letting us know. Would you mind sharing a reference for this?

nagyrobi, 2024-04-08:
https://github.com/home-assistant/core/issues/106159#issuecomment-1866786470

felipecrs, 2024-04-08:
Closing this issue as I believe @nagyrobi's [suggestion](https://github.com/esphome/feature-requests/issues/2122#issuecomment-2042987104) is good enough.

closed 2024-04-08 by felipecrs