esphome/feature-requests #2126: support for LAN8670 10Base-T1S

gkasprow, 2023-02-19:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

gkasprow, 2023-02-19:
I can build such HW configuration and write the PHY driver, the only problem I see is how to integrate it to ESPHome.

nagyrobi, 2023-10-02:
Biggest advantage: no switch needed to interconnect up to 8 devices, using a single twisted pair!!!!

nagyrobi, 2023-10-02:
https://github.com/espressif/esp-idf/issues/10799

1965z6987, 2023-10-03:
If you @nagyrobi will write driver, I can open issue in esphome. For 10BASE-T1S using lan8670, lan8671 (hw already exists, prototypes).

nagyrobi, 2023-10-04:
Please show me where to find these prototypes.

adeepn, 2023-12-06:
Driver for esp-idf https://github.com/espressif/esp-eth-drivers/pull/19

gkasprow, 2023-12-06:
how Great! But how to call it in yaml script?
[image]

adeepn, 2023-12-06:
> how Great! But how to call it in yaml script?

Perhaps it would be worth first writing support for these drivers in ESPHome?

gkasprow, 2023-12-06:
It seems so. I naively assumed that ESP just passes the argument without checking to the idf tool :)

adeepn, 2024-10-28:
> modern ESP32 like C3 or S3

For these chips https://www.microchip.com/en-us/product/LAN8650

LAN8670 we use on common ESP32

gkasprow, 2024-10-28:
my [coleague ](https://github.com/wizath/esphome) added both LAN8670 and 8650 to ESPHOME, he is currently testing it and after tests will make MR

GreenBard, 2025-05-30:
Hi all, 
I'm quite new to this project, but I'm actually looking for exactly the solution you described — a simple-to-set-up MCU with SPE Multidrop connectivity to reduce cabling.
So I'm really curious: how did it go? Did it work out? What's the current status?

adeepn, 2025-05-30:
> Did it work out? What's the current status?
it works. pre-sale status.
[image]

GreenBard, 2025-06-02:
Very nice! I'm looking forward to trying it out. One more question: What does your gateway to regular Ethernet look like? Do you use a specific switch or media converter?

adeepn, 2025-06-03:
> Very nice! I'm looking forward to trying it out. One more question: What does your gateway to regular Ethernet look like? Do you use a specific switch or media converter?

We have ethernet swotch 2x100Base-T + 1x10Base-T1S

[image]

adeepn, 2025-06-03:
> I designed a simple router based on RPI zero https://github.com/sinara-hw/SPE_router It uses two Onsemi MAC+PHY, original Ethernet library, provides also PoDL

We designed USB-adapter for debug purpose. in other cases switch covers all functions.