esphome/feature-requests #2128: Support the ESP8266_RTOS_SDK

Mat931, 2023-02-20:
**Describe the problem you have/What new integration you would like**

- The [ESP8266_NONOS_SDK](https://github.com/espressif/ESP8266_NONOS_SDK) (which is used by [ESP8266 Arduino](https://github.com/esp8266/Arduino)) no longer receives feature updates (starting from December 2019). [[Source](https://github.com/espressif/ESP8266_NONOS_SDK/blob/c6ab959ad630dc661fdf74ef8852d6501a321e0c/README.md?plain=1#L5)]
- New features like [WPA3 and PMF](https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.4) are only supported by the [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK).

**Please describe your use case for this integration and alternatives you've tried:**

- I would like to enable WPA3 and PMF for my WiFi network. Currently there are no alternatives.
- I hope the RTOS SDK will also improve the connection issues on all my ESP8266 devices.

**Additional context**

- Discussion about migrating the Arduino framework to the ESP8266_RTOS_SDK: https://github.com/esp8266/Arduino/issues/5790
- Espressif is planning to migrate the ESP8266_RTOS_SDK to esp-idf [[1](https://github.com/espressif/ESP8266_RTOS_SDK#roadmap)] [[2](https://github.com/espressif/ESP8266_RTOS_SDK/issues/1166)]

benedikt-bartscher, 2023-03-19:
Thanks @Mat931 for pointing to the new SDK. Maybe this saves me from opening all my wall switches (about 40) and spending a lot of money.
Currently i run a seperate wifi only for esphome with esp8266, but its less secure and blocks wifi channels.

Rudd-O, 2024-12-30:
This would be fantastic.  I have about two dozen devices that do not support protected management frames or WPA3.  This is fairly bad because it means a hostile neighbor can just hack into that wireless network and screw with me / my cameras.  It would be important to get this serious security hole in many ESPHome devices fixed.

ssieb, 2024-12-30:
WPA2 is not insecure.  It's not a "serious security hole".

nagyrobi, 2024-12-30:
No but a "good" friend can still cause you headaches by sending repeated deauth frames in the air preventing your nodes from reconnecting to the AP. All he needs is an ESP8266... 

Good luck troubleshooting and defeating that...

Gunni, 2024-12-30:
Please note that PMF should be possible with WPA2 too.

nagyrobi, 2024-12-30:
Afaik with upgrading the framework it's the same effort to bring PMF in along with WPA3.

Rudd-O, 2025-01-02:
8266 needs to gain WPA3.  Hundreds of thousands of people with dozens of devices each can't just have devices vulnerable to shitty encryption and deauth / reauth attacks.  Replacing those devices which otherwise should be functional and up to spec would be an enormous cost to society.

What needs to be done in order to solve the issue?  Will it require financial backing?  I would be interested in helping that way.

nagyrobi, 2025-01-07:
It's a big stab but I think most of the steps to be done should follow the path of ESP32, when ESP-IDF was introduced. So it's not really from ground zero, as at least the way that has to me made should be familiar.

It's indeed a lot of manual coding to do and thus I agree that a good financial offer to somebody willing to do it could speed up the process. Don't think that existing fulltime team members could handle it though.

Rudd-O, 2025-01-07:
I completely agree.  It's why I am willing do settle for the minimum set of features I outlined.

If anyone from the team or a volunteer with bona fides is willing to take me up on the offer, please post here.  Whatever you need in terms of $ will be considered.

The potential to move millions of existing devices into the present and future is at hand.

nagyrobi, 2025-01-20:
Chiming in @stellar-aria from https://github.com/esphome/esphome/pull/8106 might worth joining efforts, or at least consider the changes with respect to this too.

techyporcupine, 2025-05-31:
I would also be very interested in support for this. I have around 13 ESP8266 devices in my home and replacing all of them just in order to have WPA3 support would be a pain.

luckylinux, 2025-06-07:
I second what @techyporcupine said. I have so many Smart Switches (probably around 30 or even 40) with ESP8266 or even the ESP8285 (such as on the Sonoff 4Ch R3 Pro).

We simply do NOT want to force upgrade all of our Hardware (that's what Planned Obsolence would be) "just" because of this.

I am still running WPA2 only Network with a VERY strong Passphrase.

WPA3 with Compatibility Mode enabled has its own set of Issues, whether with ESP8266/ESP32 or other Devices.

KingOfNuls, 2025-09-25:
This would be great.