esphome/feature-requests #2129: Only "update all" of those devices that are actually outdated

andreas-bulling, 2023-02-21:
Hi,

With 30+ esphome devices in my HA installation I'd love to use the "Update all" functionality provided by the dashboard. Unfortunately, this updates really **all** devices - independent of their status. That is, even devices that already have the latest firmware installed are being updated. A smarter way of doing this, and much more useful and user-friendly, is to update only those devices for which a firmware update is available.

lecrams, 2023-02-23:
even better is to have a pulldown menu where you can select one or more devices to be updated. 
Using the color scheme of the dashboard while displaying the devices to be selected would make it easy to identify a device

fdcastel, 2023-06-18:
Related: #2205

TommySharpNZ, 2023-10-02:
Yeah this seems totally random to try and update devices that are already on the latest version....

ph0tic, 2025-04-28:
How is this not a thing still?