esphome/feature-requests #2130: [Components][SN74HCxx5 I/O Expander] Add possibility to work on same GPIO pins

ESP4Ever, 2023-02-21:
**Describe the problem you have/What new integration you would like**

The development boards [Dingtian Relay Development Board](https://github.com/dtlzp/relay_dev_demo) with number of relays/inputs, more than 4 are using SN74HC165 & SN74HC595 I/O Expanders, sharing same GPIO except "data" pin
Please check this 8x I/O board GPIO mapping:
[image]
The current SN74HCxx5 driver implementation doesn't allow to define them simultaneously, only OR-OR. Being defined together, none is working.

Can you please modify them to allow this scenario?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ssieb, 2023-02-21:
Which pins are shared and what do you mean by not working?  It doesn't compile or the outputs don't work?

Kentrp, 2023-06-20:
https://github.com/kecajtop/dtr0xx_io 
try this, will it help you?

ESP4Ever, 2023-12-25:
Hello!

Since update 2023.12.2 this solution can't be compiled, error:

```
.........
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

binary_sensor.gpio: [source <unicode string>:207]
  
  Pin 7 is used in multiple places.
  platform: gpio
  id: input1
  name: Input 1
  pin: 
    dtr0xx_io: dtr0xx_io_hub
    number: 7
    inverted: True
    mode: 
      input: True
      output: False
  filters: 
    - delayed_on_off: 100ms
  on_press: 
    - then: 
        - switch.toggle: 
            id: relay1
  disabled_by_default: False
binary_sensor.gpio: [source <unicode string>:221]
  
  Pin 6 is used in multiple places.
  platform: gpio
  id: input2
  name: Input 2
  pin: 
    dtr0xx_io: dtr0xx_io_hub
    number: 6
    inverted: True
    mode: 
      input: True
      output: False
  filters: 
    - delayed_on_off: 100ms
  on_press: 
    - then: 
        - switch.toggle: 
            id: relay2
  disabled_by_default: False
binary_sensor.gpio: [source <unicode string>:235] 
.........
```

Can you fix it pleeeeease ?
Related issue: [https://github.com/esphome/issues/issues/5286](https://github.com/esphome/issues/issues/5286)

Kentrp, 2023-12-28:
Try it now. It will require allow_other_uses: true flag with each dtr0xx_io component gpio and 13 pin too

allbeq, 2024-02-21:
Hi @Kentrp,
should it work with version 2023.11.6?
Actually I do not try your component before but I faced with error: **Component not found: dtr0xx_io.**
I have tried your component as local and git.. both cases I have the same issue - any ideas? 
Thanks in advance

Kentrp, 2024-02-22:
Hi! Try to clean build files in ESP dashboard and reinstall or delete (dtr0xx_io) /config/esphome/custom_components/dtr0xx_io  and install

allbeq, 2024-02-22:
Hi, unfortunately no luck with the hint, still the same ;/

allbeq, 2024-02-22:
Huh, Its working but I update node with local esphome instance and create your component as local. Thank you