esphome/feature-requests #2131: ESP_Touch for ESP32-S2 and S3

TLCary, 2023-02-22:
**Describe the problem you have/What new integration you would like**

I would like the ESP32_Touch functionality to be implemented in ESP32 S2 and S3 devices.

**Please describe your use case for this integration and alternatives you've tried:**

SOC_TOUCH_VERSION_2 is implemented in Tasmota and ESPEasy.  So I'm forced to use them for now.

**Additional context**

As of release (v12.1.1.3) S2 & S3 support the re-written touch driver: SOC_TOUCH_VERSION_2 . The default threshold for this driver is 40000. It's higher than the old driver or what is reflected in the ESPHome ESP32 Touch Pad Documentation.

AlirezaT, 2023-04-15:
I'm also looking for this!

czechmate777, 2023-04-25:
Issue posted here:
https://github.com/esphome/issues/issues/4411

kbx81, 2023-07-18:
Hi! Would love some feedback on [this PR](https://github.com/esphome/esphome/pull/5116) -- let me know if it works for any of you here. 😄

Updated documentation preview is available [here](https://deploy-preview-3077--esphome.netlify.app/components/binary_sensor/esp32_touch.html) -- note that there are some new parameters as the S2 & S3 variants have a different interface under the hood for the touch sensor hardware.

closed 2023-08-07 by jesserockz