esphome/feature-requests #2133: SPS30 Particulate Matter Sensor - implement idle mode

MatejBosansky, 2023-02-23:
**Describe the problem you have/What new integration you would like**

I was able to successfully implement SPS30 Particulate Matter Sensor. 
https://esphome.io/components/sensor/sps30.html
First point - the wiring diagram is incorrect. There is switched SEL and GND pin [(see page 4)](https://eu.mouser.com/datasheet/2/682/seri_s_a0010891602_1-2291035.pdf). But since both must be connected to GND, it didn't cause issues so far.

The problem is:
Measurement works fine. However I observed that fan inside is constantly spinning - I can feel vibrations inside sensor. I configured update_interval to 300s.
According to documentation, fan should spin only during the measurement. In idle mode fan has to be turned off.
https://sensirion.com/media/documents/188A2C3C/6166F165/Sensirion_Particulate_Matter_AppNotes_SPS30_Low_Power_Operation_D1.pdf

So it looks like sensor is constantly in measurement mode. This will definitively raise power consumption.  I am not sure but this may also reduce the sensor lifetime that is currently rated on 8-10 years with non-constant measurement mode.

According to documentation is obvious that is good when fan is spinning some time (around 15-30s) before measurement is taken.

As a solution, would be good to be able define also time how long must fan run before measurement is taken (or multiple averaged measurements are taken). If update interval is 300s and measurement interval is 30s, then it will be in measurement mode 30s and rest 270s will be in idle mode with reduced consumption.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

PlantsThatsWhatsUp, 2023-03-08:
Agreed! Would love to see idle mode implemented

thomasvnl, 2023-03-29:
I agree as well. Also, this would increase the life of the SPS30 module as the fan has increased wear on it while doing essentialy nothing useful (for the end user).

Flamingo-tech, 2023-04-01:
Agreed !

mkormout, 2023-04-13:
Yep, besides the lifespan it's noisy in continuous measurement... There is an ability to stop the measurement and start again when needed... Thanks!

tiimsvk, 2023-09-11:
https://github.com/esphome/feature-requests/issues/2385
any updates?

mabnz, 2024-02-26:
Is it possible this could be implemented?

tiimsvk, 2024-02-26:
test my own component.
It works well for me on the weather station
https://github.com/tiimsvk/SPS30-idle-mode

or SK CZ forum:
https://www.homeassistant-cz.cz/viewtopic.php?p=12644&hilit=sps30

DR-PLANTECHSTEIN, 2024-03-18:
@tiimsvk Can you submit as a PR so that it can be implemented?

c0mputerguru, 2024-05-29:
I stumbled onto this thread because my sps30 sensor died, and I think it was because the fan was running constantly for 2+ years.

I took @tiimsvk's approach, cleaned it up a bit by adding the start/stop as ations, and have a draft pull request ready: https://github.com/esphome/esphome/pull/6828

I even found a way to account for warm-up of the sensor in yaml rather than have to rewrite the whole component to have a separate thread to manage the start/stop of the sensor.

I'm waiting on a new sensor to show up so I can actually test it, and then I'll write up documentation and get this change ready to submit.  If others want to test it out, feel free to pull my changes and test them out as a local component.