esphome/feature-requests #2139: Add "Build all" button to Dashboard.

catgiggle, 2023-03-01:
**Describe the problem you have/What new integration you would like**

Adding a "Build all" button next to "Update all" in Dashboard.

**Please describe your use case for this integration and alternatives you've tried:**

I often use "Update all" because it is very convenient for me. The process takes a long time, so I can start it when I have some free time. However, I would prefer if the building process could happen without updating the device itself. I would like to have more control over it and update devices only when I think it is the right time. For example, it is not a good idea to update the lighting driver in the evening. I don't like it when the lights suddenly go out :)

Thanks to the "Build all" button, the devices will already be built, so the update will be much faster. Additionally, the interface nicely highlights devices for which a newer version of the software is available, so it would be a great addition to this mechanism.

**Additional context**

fdcastel, 2023-06-18:
Not sure if this will be need if we implement #2205 (which seems to be a better path for large ESPHome installs).

financelurker, 2023-10-31:
It would also be awesome if such an sensor/entity is exposed to HA itself so an automation could trigger the build - so when you're updating it in the UI it's just the deployment of already built firmwares.

catgiggle, 2026-09-01:
It is now possible to select any devices you want and compile or install them (especially all of them). This actually provides even more functionality than I described in the requirements of this feature request. Thank you for implementing this new functionality! I’m closing this issue as the requirements have now been met.

closed 2026-09-01 by catgiggle

catgiggle, 2026-09-01:
> It would also be awesome if such an sensor/entity is exposed to HA itself so an automation could trigger the build - so when you're updating it in the UI it's just the deployment of already built firmwares.

I agree, that would be a great addition, but I don’t think it’s really part of the scope of this issue. I think it would be worth opening a separate feature request for it. I’m definitely in favor of it. At the moment, something similar can be achieved by running your own server and providing a `manifest.json`. However, that requires quite a lot of setup and more importantly, maintenance. It makes sense for someone producing and deploying devices at scale, but it’s a lot of overhead for a typical home setup. So I’d definitely support having this functionality integrated into ESPHome/HA in the future.