esphome/feature-requests #2141: Be able to turn off, ESPhome devices showing in update tracker

PaulAntonDeen, 2023-03-02:
**Describe the problem you have/What new integration you would like**
I have ESP devices which require me to pull a panel from a cabinet, extract the ESP-12F from its relay board, plug it into my firmware burner and then write the update file to it. I'm not doing that every time a ESPHome update is released. It runs multiple relays over a serial chip, that code isn't changing much between releases. Most of my esphome devices do not need an update after they have been created, ever as they run pretty basic functions.

So I think ESPhome devices should not be part of the update notifications in core. At least I would like to be able to turn update notifications of for all ESPhome devices. I will read the release notes and decide to update them only if I see fixes or breaking changes for things I ues on my devices.

A check mark somewhere to disable esphome update notifications from showing up in the update notification at the top of the settings menu (and as a orange bubble on the cog wheel).

**Additional context**

nagyrobi, 2023-03-03:
What I did is removed the ESPHome addon and created a separate VM/LXC not tied in any way to HA. This returns to the old operation.

ronaldt80, 2023-03-18:
+1 -> to avoid doing this multiple times per month I stopped updating the add-on.. from my perspective this is not a feature but a bug:)

hareeshmu, 2023-03-24:
There should be some option to disable the update notification for the esphome nodes. Its not necessory to update the esphome sensors nodes everytime unless there is a real need.

Tozzi89, 2023-09-12:
I agree with this. Especially considered how often ESPHome is updated it is just annoying to update all small sensors that often. It would be nice to select/deselect which sensors doesn't need to update every time. Maybe a postpone button to 1 year or so?

randybb, 2023-11-16:
There already is that option, just go to esphome addon configuration and disable _Home Assistant Dashboard Integration_. But yea, an option to disable (or enable) just for some nodes could be handy.

MRobi1, 2023-11-16:
Interesting. Does this disable any other features or strictly the update notification?

randybb, 2023-11-16:
As far as I know that is only for update notification in HA. In esphome dashboard it will still show that an update is available. For now it is useful when you have multiple instances of esphome that are sharing the same directory, so only your "main" esphome (in my case dev) will pop up an update in HA.

MRobi1, 2023-11-22:
This did not work for me. 
Home Assistant Dashboard Integration was already disabled. So I enabled it. Got all the individual update notifications. Disabled it again and still getting all of the individual update notifications.

rnizametdinov, 2024-06-25:
>>>+1 -> to avoid doing this multiple times per month I stopped updating the add-on..
Me to.

mstuczko, 2024-07-21:
+1. 
My suggestion is to have it as a section into YAML, 

# no parameters, something like this:
skip_from_update_all:

# or in OTA section
ota:
  skip_from_update_all: true

For sake of visibility, some indicator may be on the dashboard.

TheHomieFox, 2024-08-10:
I found that by disabling the update entity of each ESPHome device in Home Assistant solves this issue. However, it would be nice to have a way to turn on or off the update notifications and not have to disable all those entities.

FajnB, 2024-09-05:
+1

- My motivation to implement some sort of update notification control is that not every ESP's hardware is stable and for these I go by a rule "If it works, DO NOT TOUCH IT!"
Some might even connect to home assistant on irregular basis or in very, very short intervals as they might be battery powered or have terrible connection.
- Second, and arguably the bigger, problem is that some users, my self including, have stopped updating ESPHome add-on altogether to avoid dismissing/skipping updates for potentially tenths of devices multiple times a month just because the add-on updated. It literally clutters the update center and makes it very distracting and potentially even "hides" some important updates of add-ons as they are littered like bread crumbs in the see of ESP devices.

My possible solutions are as follows:
1. Add an option toggle in ESPHome add-on setting. Something like:
```
Check for device firmware updates
Enables/Disables checking for firmware update on currently active (online) devices that are connected to Home Assistant. Disabling this option will stop creating device firmware update notifications inside Home Assistant update center.
NOTE - This option will be ignored if the update contains security patch or the device's firmware version is incompatible with the ESPHome add-on's version.
```
2. Add new yaml parameter called something like "homeassistant_update_check". It being homeassistant specific I'd add it alone or under `api:`. It's default state would be `true` and it should be present in the default yaml configuration by default so that users are made aware of it's existence.
This could also be a switch entity so that you can change it's state from Home Assistant and not hard code it.
3. Add a toggle option within Home Assistant under the ESPHome device with similar explanation as in the 1st option.

Imho I feel like some combination all above mentioned would make a robust and user friendly way to take control over these very much annoying update notifications.
Adding to my point is Home Assistant's dev team efforts to make HA as user friendly as possible.

Current solution is very much cumbersome and more of a hack/workaround than a real solution.

KazWolfe, 2025-07-04:
> NOTE - This option will be ignored if the update contains security patch or the device's firmware version is incompatible with the ESPHome add-on's version.

I think this is very important. I'm wary to disable my `update` entities because I want to make sure I don't miss a security update or something essential for smart-home operation. Otherwise though, my ESPHome devices don't need to be touched.