esphome/feature-requests #2143: Tuya Dimmer support cwww style light

clinta, 2023-03-03:
**Describe the problem you have/What new integration you would like**

I have a light using a tuya mcu, which has separate lights for cold white and warm white. This is currently implemented as two separate tuya dimmers. I'd like to be able to combine these to create a cwww light.

**Please describe your use case for this integration and alternatives you've tried:**

I've tried using the tuya dimmers as outputs to a cwww light, but that fails with the error `ID 'tuya_light_cold' of type light::LightState doesn't inherit from output::FloatOutput. Please double check your ID is pointing to the correct value.`

I've also tried creating output templates which when written to change the values of the appropriate tuya light. This works in one direction, but if I change the light via the button on the device, I cannot get the updated values back to the cwww light. When I try to use an automation to do this I end up with an infinite loop that crashes the device.

**Additional context**
My work so far is based on the excellent work of @billchurch, who documented converting a tuya celing fan/light to esphome [here](https://1projectaweek.com/blog/2022/2/8/converting-a-carro-home-dc-fan-to-esp-home)

And I'm using his [config](https://gist.github.com/billchurch/2feb716ef30f03df775da81822f59b12) as a starting point.

mgoeller, 2023-03-06:
Can you send a link to the Tuya device you are using?

clinta, 2023-03-06:
This is the device: https://www.amazon.com/dp/B0BSF224JC

The w3bl chip has been swapped out for an esp-12f.

clinta, 2023-03-14:
Here is the relevant part of the config I have now. It somewhat works, but only in one direction. If I use the rf remote that communicates directly with the tuya MCU, that change isn't brought back into the cwww light. Additionally it seems buggy with respect to brightness. Sometimes the cwww light reports full brightness, but the light is still dim. And fixing it requires turning the light off and back on.

```yaml
uart:
  tx_pin: GPIO21
  rx_pin: GPIO20
  baud_rate: 9600

tuya:

fan:
  - platform: "tuya"
    name: "Dining Room Fan"
    switch_datapoint: 1 
    speed_datapoint: 3
    direction_datapoint: 2
    speed_count: 10

light:
  - platform: "tuya"
    id: tuya_light_warm
    internal: True
    dimmer_datapoint: 10
    switch_datapoint: 9
    min_value: 100
    max_value: 1000
  - platform: "tuya"
    internal: True
    id: tuya_light_cold
    dimmer_datapoint: 102
    switch_datapoint: 101
    min_value: 100
    max_value: 1000
  - platform: cwww
    name: Dining Room Light
    id: cwww_light
    cold_white: cold_light
    warm_white: warm_light
    cold_white_color_temperature: 4000 K
    warm_white_color_temperature: 370 mireds # ~ 2700K, avoid mireds rounding errors
    default_transition_length: 0s

output:
  - platform: template
    id: warm_light
    type: float
    write_action:
      - light.control:
          id: tuya_light_warm
          brightness: !lambda return state;
          state: !lambda return state > 0;
  - platform: template
    id: cold_light
    type: float
    write_action:
      - light.control:
          id: tuya_light_cold
          brightness: !lambda return state;
          state: !lambda return state > 0;
```

richardhopton, 2023-04-04:
you can use `id(tuya_mcu).set_integer_datapoint_value(dp_id, value_int)` to set the datapoint values on the Tuya MCU directly, and `on_datapoint_update` to handle when the values change (from the RF remote?!) 

Agreed it would be nice to have this natively supported, but in the meantime, this might help you out.