esphome/feature-requests #2145: Template/!include support in ESPHome Dashboard

zendes, 2023-03-04:
**Describe the problem you have/What new integration you would like**
I'm using `<< !include file...` to have a base device config and then multiple instances of that device (ex: one tuna-dimmer-base.yaml base config and multiple livingroom/bedroom.yaml devices). Currently the *-base.yaml config also shows up as a device, which is incorrect. My proposal would be to add support to the UI to recognize these base templates and not treat them as devices but just as files. So no update button, no online/offline status etc, but still retain the editor to make changes to the files.

**Please describe your use case for this integration and alternatives you've tried:**
Support for this would make device templates easier to use. I'm aware you can create hidden files (starting with .) but that just hides them completely. You lose the ability to edit these files from the normal UI, which imo isn't a good trade-off.

**Additional context**
I'm interested in contributing this feature, but would like some feedback first on:
* is this something people are interested in?
* have people attempted this in the past and what worked/failed?
* directional guidance on how the developers would like to solve this. I can come up with my own way, but being on the same page means a higher chance of the PR succeeding. for instance: do we extend [api/devices.ts:ListDevicesResult](https://github.com/esphome/dashboard/blob/main/src/api/devices.ts#L5-L31) with a list of templates, or update [devices/configured-device-card.ts:ESPHomeConfiguredDeviceCard](https://github.com/esphome/dashboard/blob/main/src/devices/configured-device-card.ts) to hide the status/update button,

randybb, 2023-03-05:
If you think this is needed then feel free. But I don't think it is very useful - most people just use a "common" directory for all included files and since you cannot edit files in subdirectories in the dashboard, so most people use VS Code add-on (or just use vim).

fdcastel, 2023-06-18:
Put your "non-devices" files in a subfolder (e.g. `common`) from the one you are using for your devices. ESPHome will ignore them.

Alternatively, you could put a "." in front of files you don't want to appear on ESPHome dashboard.

[Here](https://esphome.io/guides/diy.html#sample-configurations) is a list of sample configurations. You may find several tricks useful.

zendes, 2023-07-17:
sounds like most people are working around this issue in other ways and are happy with how it performs for them. i'll probably do the same then. thanks all for the feedback

closed 2023-07-17 by zendes