esphome/feature-requests #2146: Enter text in text_sensor component via web component

matsekberg, 2023-03-05:
**Describe the problem you have/What new integration you would like**
If using esphome stand-alone there is sometimes a need to enter configuration data such as api keys. A text_sensor component rendered as a text input box in the web gui would be perfect.

**Please describe your use case for this integration and alternatives you've tried:**
Imagine an esphome with a generic config, connected to Wi-Fi and a temp sensor and reporting temp to a remote server.
When starting the device for the first time the captive portal will enable the user to enter Wi-Fi config.
The the user can surf to the device an enter some config like API-key to remote site in a text input box in the web gui.

**Additional context**

oxan, 2023-03-08:
This would need support for the new text domain from home assistant, it's not something that fits with the text_sensor.

Samu84, 2023-03-14:
Hello. yes this it would be perfect, this is something i am looking all day today,  for example it would be good to be able to put the parameter of the MQTT user, password, etc... (of course not everyting in one box) or modify the word on the display, using the web interface  and not writing all the time the bin file.
for example in this picture, that is on the official site, there are, the option where you can write the numbers:
6 digits:
template number:
I am already using template number  ( unfortunately, i only have the option with line like in AC, if someone know how to do it, it would be nice).
However going back to write the text like the number it would be very usefull

[image]

Samu84, 2023-03-24:
number box i finally found a way, thanks to #[ssieb](https://github.com/ssieb).
Hope there will be a way for the Text box

Jan3726, 2023-08-14:
Hello, this is something else i am interested too, it would be nice to write a text or number starting with 0 (for example binary number) in to the text box

EternityForest, 2023-08-22:
@Jan3726 it seems there is an ongoing PR: https://github.com/esphome/esphome/pull/3638

I still wonder if a specialized web configuration feature wouldn't be better, so that your "Settings" could stay separate from your "runtime values", if you had a few dozen config options you'd at least want some sort of flag to stash them, and also just to not overload the text components with extra config UI specific stuff.

Doing everything through just the one text component does seem logical though.

EternityForest, 2023-10-25:
I believe this use case is covered by this new merge!

https://github.com/esphome/esphome/pull/5336

nagyrobi, 2023-10-25:
Not sure if it appears in ESPHome's web UI, it's rather usable through HA I think.

EternityForest, 2023-10-25:
It's available in web server v2, I don't think I ever did anything in v1 though.

https://github.com/esphome/esphome-webserver/pull/43#event-10761489030