esphome/feature-requests #2152: Support airwell for climate IR

sebakerckhof, 2023-03-07:
**Describe the problem you have/What new integration you would like**

I have an airwell HVAC, which isn't yet supported by the https://esphome.io/components/climate/climate_ir.html component.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

It's supported by tasmota which uses the https://github.com/crankyoldgit/IRremoteESP8266 library, so the codes are already decoded

nagyrobi, 2023-03-07:
Some Airwells are actually Gree clones. A Gree component is [being added](https://github.com/esphome/esphome/pull/4464) to ESPHome. My Airwells work with:

```
external_components:
  - source: github://pr#4464
    components: 
      - gree
    refresh: 5min

climate:
- platform: gree
  name: ${friendly_name} klíma teszt
  sensor: current_temperature
#  model: generic #not OK
#  model: yan #OK
#  model: yaa # OK
  model: yac #OK

sensor:
  - platform: homeassistant
    id: current_temperature
    entity_id: sensor.xiaomi_ble_temp
    state_class: measurement
    unit_of_measurement: "°C"
    accuracy_decimals: 2
    internal: false
    filters:
      - quantile:
          window_size: 7
          send_every: 3
          quantile: .3
      - delta: 0.03
      - throttle: 15s
```

mdrwiega, 2024-03-22:
@nagyrobi Which Airwell model do you have? Have you managed to integrate it with Home Assistant?

nagyrobi, 2024-03-22:
airwell awsi-hdd012-n11

closed 2024-06-22 by nagyrobi