esphome/feature-requests #2153: Climate Zones

clinta, 2023-03-07:
**Describe the problem you have/What new integration you would like**

I would like to be able to associate multiple climate thermostats together, so that `run_time` and `off_time` timers are shared between the thermostats, and `deadband` settings are ignored when heating or cooling is active on any one of the thermostats.

**Please describe your use case for this integration and alternatives you've tried:**

When using a multi-zone heating system, it makes sense to have a climate thermostat for each zones. But some settings should apply to the entire heating system, not to the zones. Like `min_heating_run_time` and `min_heating_off_time`, these are to protect the heating system and prevent short cycling. But once the heating system is on, because one zone is on, the as long as all zones don't turn off before min_heating_run_time, individual zones do not need to stay on for that duration. Additionally, once a zone is on, `heat_deadband` can be ignored. Once the system is running, it would be beneficial to ignore the deadband and start heating every zone up to `heat_overrun`, again to minimize the startup cycles of the heating system and avoid the case of one zone reaching temperature, and shutting off the heater right before another zone passes the deadband and starts calling for heat.

**Additional context**

nagyrobi, 2023-03-07:
I'm in the same boat! I have 7 zones (different rooms). Here's how I try to deal with this now:

A thermostat for each room:

```yaml
- platform: thermostat
  name: ${zone_1_name} heating
  id: heat_01_term
  sensor: heat_01_temp
  startup_delay: false
  default_preset: heat
  on_boot_restore_from: default_preset
  preset:
    - name: heat
      default_target_temperature_low: ${default_target_temp}
  visual:
    min_temperature: 14
    max_temperature: 25
    temperature_step: 0.1
  min_idle_time: 5min
  min_heating_off_time: 7min
  min_heating_run_time: 7min
  heat_deadband: 0.2 °C
  heat_overrun:  0.2 °C
  heat_action:
    - switch.turn_on: heat_01_sw
  idle_action:
    - switch.turn_off: heat_01_sw
```

Switch `heat_01_sw` is a relay commanding one or more TRV valves allowing hot water to flow in the radiators corresponding to that room. Time needed for a TRV to open is about 2-3 minutes.
There's a separate relay commanding the central heating boiler, `heater_sw`:

```yaml
### Big OR switch between the zones
- platform: template
  entity_category: diagnostic
  name: "Heat demanded"
  icon: mdi:gas-burner
  lambda: |-
      return (
        (id(heat_01_sw).state or
        id(heat_02_sw).state or
        id(heat_03_sw).state or
        id(heat_04_sw).state or
        id(heat_05_sw).state or
        id(heat_06_sw).state or
        id(heat_07_sw).state) or
      );
  on_state:
    - lambda: |-
        if (x) {
          id(heater_sw).publish_state(true);
        } else {
          id(heater_sw).publish_state(false);
          }
        }
### Gas burner debounce+delay, to protect
- platform: template
  id: heater_sw
  filters:
    - delayed_on_off: 20s
  on_state:
    - lambda: |-
        if (x) {
          id(heater_relay).turn_on();
        } else {
          id(heater_relay).turn_off();
        }
```

[image]

clinta, 2023-03-07:
It would be a little messy, but I could add the functionality I want with automatons if the `climate.control` action could dynamically change the `heat_deadband` and `min_heating_run_time` values.

nagyrobi, 2023-03-07:
Or if they were templatable?

clinta, 2023-08-13:
Thinking about winter once again, and I discovered I may be able to achieve this with lambdas using `set_heat_deadband` and `set_heating_minimum_runtime_in_sec`. It'll be complicated, but if I get something working I'll update here.

https://esphome.io/api/classesphome_1_1thermostat_1_1_thermostat_climate

Syka2210, 2023-11-21:
Any luck with resolving the short cycling?