esphome/feature-requests #2156: Add support for LD2410 BLE photosensitivity

holdestmade, 2023-03-08:
**Describe the problem you have/What new integration you would like**
Add support for LD2410 BLE photosensitivity

**Please describe your use case for this integration and alternatives you've tried:**
Firmware 2.4 adds photo sensor, can this be added to Esphome ?

**Additional context**

regevbr, 2023-03-13:
I tried to get the serial protocol documentation for that version from the manufacturer, but they don't have it yet. Once I get it (or anyone else) I will be more than happy to implement it.

user45876, 2023-03-17:
From inside the HLK-App, a more current versdion of the documentation can be opened (uppermost left button when scanning for modules). It is version 1.06. I have google-translated the relevant pages 16-20. They do mention the light sensitivity function, however I am unsure if it is possible to extract a value or only if it is above or below a threshold, which can be set 1-255. 

Please find the pages attached and thanks a lot for your efforts :) https://imgur.com/a/PHX0FCV

Extract from page 16:

```
2.2.18 Auxiliary control function setting
This module comes with a photosensitive diode, which can be used to detect the output light sensitivity value (please refer to), and the user can also configure the light sensor auxiliary control function;
Turn on the light-sensing auxiliary control function, the output of the OUT pin is affected by the radar detection result and the light-sensing control logic at the same time:
The output of the OUT pin changes from no one to someone, which needs to be met: the radar detects people and the logic conditions of the light-sensing auxiliary control are met;
The output of the OUT pin changes from someone to no one, which needs to be met: the radar detects no one;
The light sensing control logic can choose to detect that the light sensing value is less than the set light sensing threshold, or to detect that the light sensing value is greater than the set light sensing threshold;
The default output level of the OUT pin can also be configured;
Command word: 0x00AD
Command value: 4 bytes configuration value
Return value: 2 bytes ACK status (0 success, failure)
Table 9 Command Values ​​for Auxiliary Control Function Settings
first byte
illustrate
0x00
Turn off the light sensor auxiliary control function, the output of the OUT pin will not be affected by the light sensor
Turn on the light-sensing auxiliary control function, and when the detected light-sensing value is less than the set threshold, the auxiliary control conditions are met

The second byte is the light sensitivity threshold to be set (range 0x00~0xFF)
Turn on the light-sensing auxiliary control function, when the light-sensing detection value is greater than the set threshold, the auxiliary control condition is satisfied; the second byte is the light-sensing threshold to be set (range 0x00~0xFF)

The factory default value is 0x00, that is, turn off the light sensor auxiliary control function
Page 16 of 18
```

Roofcatbusy, 2023-03-17:
Hello, as I see from the doc above (thanks for it) and from app interface photosensor only affects OUT pin behaviour. I doubt it is possible to extract measured or raw value of light intensity, sensor can only be used as a gate to allow or suppress OUT pin signal. I suppose that the usecase is smtg like - set OUT pin high only during night to use it as nightlight sensor.

user45876, 2023-03-17:
Indeed. The first sentence is a little unclear, however as it is hinting at the possibility to somehow 'detect the output light sensitivity value, AND [...] auxiliary control function'. 
The app is somehow showing 'light sensitivity' in engineering mode, it is the last graph. It is only changing slightly, when covering the sensor or shining light on it.

regevbr, 2023-03-17:
Thanks for the find!
I managed to extract the full pdf from the android apk and translate it.
So the light sensitivity only has an effect on the out pin. We can get the current light value when in engineering mode - 
[image]

That means I can add all the functionality to control everything and get the light but to get a constant value of the light sensor will incur the cost of being in engineering mode constantly. If you don't monitor any of the other sensors that work during engineering mode, the "cost" can be reduced.  @sebcaps and other, any thoughts?

regevbr, 2023-03-17:
What we can do thought, is set up a https://esphome.io/components/binary_sensor/gpio.html gpio binary sensor so we can monitor the out pin and use that for automation...
In any case I'm almost done adding it to the current PR

sebcaps, 2023-03-17:
In my case, as I don't have the last version I added a bh1750 lux sensor to condition automation trigger. Indeed using gpio (if correlated to light ) seems a better option rather than enabling engineering mode for constant usage. Nevertheless getting the value for fine tuning would be great.
@regevbr I don't know if it's possible to manage hardware révision in the component but could be a good thing ? ( Eg don't try to get value if HW does not support....)

regevbr, 2023-03-17:
I'm actually not sure it only exists in the newer version. Once I publish the feature in a few hours, could you please try it out and let me know if it works?

user45876, 2023-03-17:
> In my case, as I don't have the last version I added a bh1750 lux sensor to condition automation trigger. Indeed using gpio (if correlated to light ) seems a better option rather than enabling [...]

Having the value would be great. Is there a significant downtime to enabling engineering mode / having it enabled?

regevbr, 2023-03-17:
no downtime but it will create a lot more "stress" on the esp device and HA sensors as it produces a lot of changing values very fast. Not sure it will hold a reboot, so you might need to monitor the engineering mode value and always turn it on if it becomes off. Also, it seems that the light sensor is pretty shitty so I would not relay on it... bh1750 which @sebcaps and I are using is very recommended and very cheap.

user45876, 2023-03-17:
Thanks alot for your insights. I understand and ordered some right away 😁 So I just add it to i2c and add the component and some entities to the esp. Oh man, I like esphome so much 😘

regevbr, 2023-03-17:
> Thanks alot for your insights. I understand and ordered some right away grin So I just add it to i2c and add the component and some entities to the esp. Oh man, I like esphome so much kissing_heart

haha yes indeed an awsome project!

This is my config for the light sensor:

```yaml
i2c:
  frequency: 400kHz

sensor:
  - platform: bh1750
    name: "Illuminance"
    update_interval: 5s
    device_class: "illuminance"
    unit_of_measurement: "lx"
```

regevbr, 2023-03-17:
@holdestmade this feature can now be tested at - https://github.com/esphome/esphome/pull/4434#issuecomment-1474437291
I would appreciate everyone's feedback :-)

regevbr, 2023-03-22:
> I'm actually not sure it only exists in the newer version. Once I publish the feature in a few hours, could you please try it out and let me know if it works?

@sebcaps did you have a chance to give it a try?

sebcaps, 2023-03-23:
Hi, this is in progress, as well as PR review. Very astonishingly (to me) seems that values can be retrieved from an 'old' version of the device as well ..

regevbr, 2023-03-23:
I think it was always there, they just didn't document it and added it to the app. But that is good to hear it exists in "old" versions

mikesmitty, 2023-09-09:
Since support for the light sensor was added in 2023.8.0 this can probably be closed

closed 2023-09-10 by holdestmade