esphome/feature-requests #2159: Growatt sensor  - export limit add on.

StarChildDK, 2023-03-10:
**Describe the problem you have/What new integration you would like**
The growatt sensor integration - On some of the growatt inverters it is possible to set Export limitations via RS485, it would be great if esphome could control this.

**Please describe your use case for this integration and alternatives you've tried:**
Home assistant could be used to control the export limit, it would be very convenient if the home assistant already has a smart meter. 

**Additional context**

kiss81, 2023-05-27:
This Would be a really nice way to be able to turn off the output when prices are negative.

Saentist, 2023-07-08:
In **Growatt Inverter Modbus RTU Protocol V1.24** there is a option to do this:
[image]
[image]

kiss81, 2023-07-08:
> In **Growatt Inverter Modbus RTU Protocol V1.24** there is a option to do this: [image] [image]

Good find! I have no clue yet how to implement this :) Is there another project using this so I can have a look how it's implemented?

wauswaus, 2023-12-16:
We see more and more negative Kwh pricing so it would be great to automate the invertor to produce power.
Another option for this I came across is:

[image]
Could this be exposes to HA as a switch or so? Maybe more easier to implement?

Saentist, 2024-01-14:
> We see more and more negative Kwh pricing so it would be great to automate the invertor to stop produce power. Another option for this I came across is:
> 
> ![afbeelding](https://private-user-images.githubusercontent.com/70056029/290981243-6d28379e-e5b1-4a0e-a830-7e74f014005a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyNjI3NjgsIm5iZiI6MTcwNTI2MjQ2OCwicGF0aCI6Ii83MDA1NjAyOS8yOTA5ODEyNDMtNmQyODM3OWUtZTViMS00YTBlLWE4MzAtN2U3NGYwMTQwMDVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE0VDIwMDEwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjMWE3MTdiNmQ0ZTQ1ZjIzODYxZmFmZmJiNDBmMTFkYWJkMTU1YzY5OGMxNDI0YzE4OThiODQ1YWU4YTI2ZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Q-_suNzaYZtisu7BDy1SQt34WRlkeko6y5DZz3i6FiY) Could this be exposes to HA as a switch or so? Maybe more easier to implement?

Why to stop produce power? 
Idea is to be independent from electric company as much is possible,
witch mean ZERO import and ZERO export.

apromix, 2024-02-21:
Hi.

it is possible to use another register to control Export limit, namely Holding Register `3`:

[image]

I thing, to use the registers `122` and then `123` will work only via separately RS485 or RS232 port, not via USB port, so the ESPHOME must be connected two times (USB + RSxxx) and emulates the power meter (but not sure, maybe not?).

Already exists a [PR](https://github.com/OpenInverterGateway/OpenInverterGateway/pull/271/commits/ab799f81898b3a87157b3466ce4486c697d9a06f#diff-972cac2b8a5e3d6e89bd0cdce7924a5acf99e9468383e29915ac28a40e6a1ba7R354) in another project.

I already tested it and seems to be work:

Example (OpenInverterGateway project) with very small PV with 100Wp and Growatt 600tl-x:

```
$ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=10"; echo
Wrote Holding Register 3 to a value of 10!

$ curl -X POST http://growatt/postCommunicationModbus_p -d "val=&type=16b&operation=R&registerType=H&reg=3"; echo
Read 16b Holding register 3 with value 10

$ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=5"; echo
Wrote Holding Register 3 to a value of 5!

$ curl -X POST http://growatt/postCommunicationModbus_p -d "val=&type=16b&operation=R&registerType=H&reg=3"; echo
Read 16b Holding register 3 with value 5

$ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=100"; echo
Wrote Holding Register 3 to a value of 100!
```

[image]

Legend:

- [9:48:35] 10% => ca. 60W
- [9:49:00] 5% => ca. 30W
- [9:49:40] 100% => back to max

Maybe this should be the correct way ..

Saentist, 2024-02-24:
> Hi.
>
> it is possible to use another register to control Export limit, namely Holding Register `3`:
>
>
> I thing, to use the registers `122` and then `123` will work only via separately RS485 or RS232 [...]

This function wis more interesting in case if there is no power limiter installed at all
[image]

Negative Red leaf on inverter but works

Sicos1977, 2024-06-16:
> > Hi.
> > it is possible to use another register to control Export limit, namely Holding Register `3`:
> > I thing, to use the registers `122` and then `123` will work only via separately RS485 or RS232 port, not via USB port, so the ESPHOME must be connected two times (USB + RSxxx) and emulates the power meter (but not sure, maybe not?).
> > Already exists a [PR](https://github.com/OpenInverterGateway/OpenInverterGateway/pull/271/commits/ab799f81898b3a87157b3466ce4486c697d9a06f#diff-972cac2b8a5e3d6e89bd0cdce7924a5acf99e9468383e29915ac28a40e6a1ba7R354) in another project.
> > I already tested it and seems to be work:
> > Example (OpenInverterGateway project) with very small PV with 100Wp and Growatt 600tl-x:
> > ```
> > $ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=10"; echo
> > Wrote Holding Register 3 to a value of 10!
> > 
> > $ curl -X POST http://growatt/postCommunicationModbus_p -d "val=&type=16b&operation=R&registerType=H&reg=3"; echo
> > Read 16b Holding register 3 with value 10
> > 
> > $ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=5"; echo
> > Wrote Holding Register 3 to a value of 5!
> > 
> > $ curl -X POST http://growatt/postCommunicationModbus_p -d "val=&type=16b&operation=R&registerType=H&reg=3"; echo
> > Read 16b Holding register 3 with value 5
> > 
> > $ curl -X POST http://growatt/postCommunicationModbus_p -d "type=16b&operation=W&registerType=H&reg=3&val=100"; echo
> > Wrote Holding Register 3 to a value of 100!
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Legend:
> > 
> > * [9:48:35] 10% => ca. 60W
> > * [9:49:00] 5% => ca. 30W
> > * [9:49:40] 100% => back to max
> > 
> > Maybe this should be the correct way ..
> 
> This function is more interesting in case if there is no power limiter installed at all ![image](https://private-user-images.githubusercontent.com/16975883/307511782-0735e63f-90ed-41b6-9266-a80eeee88d46.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1MzE4NzIsIm5iZiI6MTcxODUzMTU3MiwicGF0aCI6Ii8xNjk3NTg4My8zMDc1MTE3ODItMDczNWU2M2YtOTBlZC00MWI2LTkyNjYtYTgwZWVlZTg4ZDQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE2VDA5NTI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIyYjcyYjllNzNjOGFhYjlhMTI3ZDI5OTQ4NDdhYTNhNjlhYTlhYzY5ODMzNThlZDdmMTE3NDk0YTM2YTM5NTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IpRgQg9Tf9H7PuY4hDEBFokiJ8Tx4skzZvSXhJDgbmM)
> 
> Negative Red leaf on inverter but works

Do you know what modbus register this is? Default Power After Exportlimit Failure?

Sicos1977, 2024-06-16:
When I set this value to 10% on my inverter that has a max output of 6000wp it wil limit the output to 600wp. So for everyone who reads this in the near feature. The value that you set here is not the percentage of the max output your panels are generating at that moment but the max your pv inverter can do.

Sicos1977, 2024-06-16:
So to control the output limit **without** a Growatt Smart meter you set the following modbus values:

- 122 to 1 = Meter 1
- 123 to 1000 = Set output limit to 100%

And then set the **Default Power After Exportlimit failure** to the following percentage

The max output of your PV inverter (in my case 6000) * 10% = limit the power output to 600 watt

I still need to find out what modbus register you need to set this percentage.

fbloemhof, 2024-08-27:
Tried implementing the code from @drahdiwaberl. It seems to work, I can limit the output state on both of my Growatt inverters.

Noticing weird behavior on one of the inverters, he limit value resets back to 100 after a minute or so, but the limit stays active. I can restore it by setting it to another value and immediate to 100.

drahdiwaberl, 2024-08-29:
> Noticing weird behavior on one of the inverters, he limit value resets back to 100 after a minute or so, but the limit stays active. I can restore it by setting it to another value and immediate to 100.

seeing this _every now and then_ either - mostly in the morning - but haven't bothered debugging yet. is it just the slider/number-entity or the sensor as well? didn't have the latter on the dashboard until now..