esphome/feature-requests #2161: Allow the use of multiple pn532_i2c RFID readers

bc547, 2023-03-10:
**Describe the problem you have/What new integration you would like**

I would like to use multiple pn532_i2c devices on a single esp32 device to implement a game.
At the moment the pn532_i2c component only allows a single reader device and it is not possible to define multiple devices.

**Please describe your use case for this integration and alternatives you've tried:**

A single device works as expected:
```
pn532_i2c:
   id: rfid1
   update_interval: 1s
   i2c_id: i2c0
```

Defining multiple devices returns an error `Failed config: expected a dictionary`
```
pn532_i2c:
  - id: rfid1
    update_interval: 1s
    i2c_id: i2c0
  - id: rfid2
    update_interval: 1s
    i2c_id: i2c1
```

**Additional context**

When I look at https://github.com/esphome/esphome/pull/3989, it seems adding the requested feature might seem rather trivial?

therealmrfox, 2024-02-04:
I am happy to report that this works now :)

lindsaymarkward, 2025-01-25:
Should this issue be closed, if this works?
Also, can the documentation be updated to show how to configure multiple devices please?