esphome/feature-requests #2162: Add support for DFPlayer Pro

Brabouk, 2023-03-11:
**Describe the problem you have/What new integration you would like**

Please add Dfplayer PRO integration from DFROBOT.

**Please describe your use case for this integration and alternatives you've tried:**

Dfplayer PRO has different commands for control via UART than dfplayer mini, which has esphome integration.

**Additional context**

I tried to make a custom switch with C++ code, which should send a command via UART, which is controlled by the PRO version, but it doesn't work for me. I'm not a good C++ programmer, so there's definitely a mistake, but I believe that with your help, dfplayer pro could also be controlled with these custom switches.

uart:
  id: my_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 115200

switch:
  - platform: custom
    lambda: |-
      auto my_switch = new Switch();
      auto uart = App.make<UARTComponent>("my_uart");
      my_switch->add_on_state_callback([&](bool state) {
        if (state) {
          uart->write("AT+VOL=5\r\n");
        }
      });
      return my_switch;

    switches:
      name: "My Custom DF PRO Switches"
      id: my_switch

hell1337hound, 2024-05-25:
Second for this!

Old-Simon-A, 2024-10-01:
I bought two DFPlayer Pro modules from DigiKey this month. My application is to have SINGLE playback. I used “AT+PLAYMODE=3” to set the play mode, and used “AT+PLAYMODE=?” to verify. It returned “3”. However, when I used it in my application, it played repeatedly. I checked it’s play mode again, and it returned “2”. Both modules have the same issue. 
When I insert the DFPlayer Pro module into my laptop, it is just like a USB memory stick that I can move files in and out of. However, when I pushed the button on the module to play the music, it became unrecognizable to my laptop. The module seems not to have a “reset“ mechanism. Does there have any way to make it return to its initial mode (i.e., the mode when it powers on)?
Get no response from DFROBOT!