esphome/feature-requests #2163: Support removing components defined in packages

LordMike, 2023-03-12:
**Describe the problem you have/What new integration you would like**
We can extend components defined in packages like this:

```yaml
packages:
  athom.rgbct-light: github://athom-tech/athom-configs/athom-rgbct-light.yaml

button:
  - id: !extend Reset
    name: Overriden name
```

But I'd like to use a package, _and_ be able to _remove_ a component. I imagine something like this:

```
button:
  - id: !remove Reset
```

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to use the packages defined by the vendor of my device, in this case Athom Tech. But sometimes they define odd components that I'd like to not have, but I can't remove them. 

A workaround is to disable or hide the component. In this case, with button, I can mark it as internal. I think the best solution would be to completely remove it from the configuration, which potentially could mean that "button" wasn't even compiled in to the final firmware. At the minimum, it's not managed in memory.

```yaml
button:
  - id: !extend Reset
    internal: true
```

**Additional context**
N/A

cooperised, 2023-06-01:
+1.  This would also be useful for overriding parts of a package configuration that conflict with local requirements without forking the package, for example to disable `esp32_improv` in order to allow `esp32_ble_beacon` to be configured.

chatziko, 2023-06-24:
+1. Another use-case to consider: I have a package with the core config of all my nodes, which includes:
```
captive_portal:
```

In just one node I want to use `esp-idf`, which however is not compatible with captive portal. So I'd like to be able to "unset" the `captive_portal` field set in the package.

closed 2023-12-06 by jesserockz