esphome/feature-requests #2164: Dashboard: Add support for creating configs that can be included in device configs

LordMike, 2023-03-12:
**Describe the problem you have/What new integration you would like**
When reading the esphome docs, I come across topics like packages and includes, which all help with managing configs for many different devices. My issue is then, that to use this with the esphome addon, I need to bring in either a directory on my HASS server which I volume mount in to the esphome docker container, or a git repository which may or may not have to be public. In both cases, the experience is not seamless.

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to have an easy way to reuse configuration across devices. Right now, I use packages from the vendors, but I'd like to also add in my own tweaks or reusable configurations.

I imagine that this would add series of UI elements that allow me to list/create/edit/remove reusable configs. Each reusable configuration would be identified by a name, which is unique. This name can then be included in my device configs, either using:

```yaml
packages:
  wifi_config: !include reusable/wifi_config.yaml
  lights_override: !include reusable/athom_light_overrides.yaml
```

**Additional context**
I don't know if it should be possible to make hierarchies of config files (maybe their name can simply include slashes to indicate directories). I also don't know if it should be possible to nest includes, as in an included config can include another config.

randybb, 2023-03-12:
As responded multiple times before, fell free to implement it. But I don't think it is needed. For simple configs, or when common files are already present and you are only including them, is the dashboard enough. For more advanced stuff we have VSCode, or you can use any other editor of your choice. Why wasting time with building our specific IDE when a more universal IDE is already available?

LordMike, 2023-03-12:
Ooh.. Shoot. I'd completely missed that the configs were in the HASS config folder - which I can see in vs code. Neat.

Sorry for the disturbance.. :)

closed 2023-03-12 by LordMike