esphome/feature-requests #2165: squeezelite support

nagyrobi, 2023-03-12:
**Describe the problem you have/What new integration you would like**
Integrate the Logitech Media Server part from Squeezelite-ESP32: https://github.com/sle118/squeezelite-esp32
Squeezelite-ESP32 project has lots of features on its own (like support for display, ethernet, buttons, gpio expanders etc), but the core part is only what we need here, which communicates with the Logitech Media Server.

This is a well-proven stable solution for multiroom audio in perfect sync, and lots of plugins allowing Spotiry, LastFM etc integration. It also integrates very well in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**
What the Squeezelite-ESP32 is missing is a way to integrate with Home Assistant for automating stuff like turning on the TV/Speakers, choose the correct audio source, etc.

**Additional context**
N/A

juanitoddd, 2023-11-16:
This would be pure awesome indeed ! 
I wonder how dificult would be to port the Squeezlite core part, which i guess is here ([https://github.com/sle118/squeezelite-esp32/tree/master-v4.3/components/squeezelite](https://github.com/sle118/squeezelite-esp32/tree/master-v4.3/components/squeezelite)) and create a squeeze runtime in ESPHome 🤔

DavidSpivey, 2025-08-22:
I'm adding a comment to "bump" this and let the devs know WE REALLY WANT THIS. I have also been using squeezelite-esp32, but having ESPHome on these ESP32 Audio Kits would enable all of them (I expect to have one in each room for whole home audio) to use the GPIO pins and make these things a powerhouse. Additionally, making these all available for assistant functions would blow the existing squeezelite implementation out of the water. On top of all that, each of these would become a bluetooth proxy, which I have been interested in for a while. Long story short, these devices that I have originally purchased for a single purpose of whole home audio would suddenly become incredibly useful as multifunction devices.

Davis8483, 2025-08-22:
@DavidSpivey I think the devs are instead working on a dedicated protocol to bring multi-room sync to esp home devices.

https://github.com/esphome/backlog/issues/20

jose1711, 2026-11-08:
FWIW, I put together a Claude-authored external component that adds basic support for the Squeezebox protocol (SlimProto) to ESPHome: https://github.com/jose1711/esphome-slimproto

It registers as a `media_source` for `speaker_source`, so I2S, the mixer and the resampler stay standard ESPHome components. I've tested it successfully on a Home Assistant Voice PE (ESPHome 2026.10) against Lyrion Music Server:

- music and radio from the LMS add-ons, and podcasts (FLAC/MP3/PCM)
- it integrates well with the wake word: playback ducks while the assistant answers
- volume stays in sync in both directions (Voice PE dial ↔ LMS)
- pausing playback by voice command works

Still experimental: multi-player sync is only basic and it hasn't been tested on other boards. Feedback welcome.