esphome/feature-requests #2166: Support for M5Stack Tough ESP32 IoT Development Board Kit

303Bryan, 2023-03-13:
[image]
**Describe the problem you have/What new integration you would like**

I would like to see support for the [M5Stack Tough ESP32 IoT Development Board Kit](https://shop.m5stack.com/collections/m5-controllers/products/m5stack-tough-esp32-iot-development-board-kit). This device uses a different hardware configuration from other M5Stack productions.

**Please describe your use case for this integration and alternatives you've tried:**

The M5Stack Tough has the potential for novel information display and home assistant control applications. The "tough" nature provides an interesting use case in environments where typical devices are not suited (for example, outdoors). Due to the differences in hardware implementation (particularly the screen), there is not an easy solution to integrate the device with Home Assistant/ESPHome. 

**Additional context**

Detailed documentation: https://docs.m5stack.com/en/core/tough

There are several threads on various forums where other M5Stack products are being worked on, but I could not find much movement for the M5 Tough yet.

RASBR, 2023-05-10:
Hi,
I add my voice to adding the support directly into ESPHome, hopefully sooner than later.

danielscheidler, 2023-10-04:
Ok, june is over, but here is my first CC for the tough.
only to put some buttons on it to trigger HA automations, but perhaps it helps anyone or someone like to extend it.

https://github.com/SmartHome-yourself/m5-tough-/tree/main

nagyrobi, 2023-10-04:
Dead link

RASBR, 2023-10-13:
Hi, @danielscheidler 
Thanks for your link. I have reached your website and sent you an email there. I would appreciate if you can check it.
Regards,
RS