esphome/feature-requests #2168: Suppress Strapping Pin Warning

clinta, 2023-03-14:
**Describe the problem you have/What new integration you would like**

I'd like a configuration option on a pin to disable compiler warnings about strapping pins.

**Please describe your use case for this integration and alternatives you've tried:**

When compiling code that uses strapping pins a compiler warning is issued

```
WARNING GPIO8 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
```

**Additional context**

As noted in the FAQ "It’s recommended to avoid them unless you have a pressing need to use them and you have reviewed the expected boot voltage levels of these pins from the ESP datasheet."

This is the case for me. This is a tuya device which I've done a chip replacement on, replacing the tuya chip with an esp-c3-12f. I can't change which pins are used. And I've tested and confirmed that the boot voltage on these pins does not cause an issue. So for this device I'd like to suppress that warning. Having it as a config option is valuable because it would warn me if I inadvertently use one of these pins in a future project. But once I've confirmed that it's not an issue on a particular device, I'd like to disable the warning.

jordan875, 2023-04-13:
+1

I have the same issue on Sonoff devices. Various Sonoff devices use GPIO0 for the button on the enclosure. This can be used to put the device into flashing mode on startup, but also serves as the relay toggle after startup. Annoyingly, because this shows up as a warning in esphome, the status led is forever flashing, which makes it impossible to know whether there is a new error.

ssieb, 2023-04-13:
The compile time warning has nothing to do with the status led flashing.

jordan875, 2023-04-13:
The warning is there at all times, not only compile time. If I open the device logs, the warning is there. According to the docs, status led will flash whenever there is a warning active. 

See [https://esphome.io/components/status_led.html](https://esphome.io/components/status_led.html)

nagyrobi, 2023-04-13:
No. That yellow text warning is only applicable before compiling. It's displayed by the config file parser. During runtime, there's no warning related to the configuration. That status LED will not be flashing because of that.

jordan875, 2023-04-13:
I stand corrected. Thank you.

ssieb, 2023-04-13:
@nagyrobi when the dashboard parses the yaml to find out how to contact the device, it probably does display the warning as well.  But that still doesn't affect anything on the device itself.

nagyrobi, 2023-04-13:
Exactly.

jordan875, 2023-04-13:
@ssieb thank you for the clarification :)

Unfocused, 2023-04-25:
This has been bugging me too. I have several devkits that use these pins internally, as well as full products that I'm re-flashing (coincidentally, also Tuya modules) - the pins are not something I can do anything about. 

The consumer products I have that are shipped with ESPHome (Athom Smart Plugs) don't use strapping pins, but I can see it being particularly important in that case: where a manufacturer sells a product running ESPHome, and provides the source for the ESPHome config. A warning like this is likely to result in a fair number of support queries with people asking what's wrong.

The warning is really useful when you have control over which pins to use & how to use them. But in other scenarios, like these, it's unwanted & distracting noise. And there's a good reason why compilers & linters generally allow inline overrides of specific warnings.

Rather than a global option to disable the warning completely, I'd prefer to be able to disable the warning for each pin individually. That'd let you still have the safety warning for any future pin use - which would be especially important for devkits and consumer products.

I see two ways this could be done:
- A global option under `esphome:` that accepts a list of safe pins:
  ```yaml
  esphome:
    disable_warnings:
      - strapping_pins: 1,2,3
  ```	
- Add it to the pin schema:
  ```yaml
  pin:
    number: GPIO01
    inverted: true
    mode:
      input: true
      pullup: true
	disable_warnings:
      - strapping
  ```
My preference would be adding it to the pin schema. It's contextual, and would also work best when using packages. Conveniently, it looks like it would be the easiest to implement too.

In those examples, I'm disabling a specific warning. ESPHome also has some other warnings, such as extra pins that might be used for Qual SPI flash. In the future, additional warnings may be added. Personally, I'd prefer to have it work like this, the way most compilers/linters allow inline disabling of warnings, and have to explicitly disable each warning individually.

The alternative would be disabling all warnings:
  ```yaml
  pin:
    number: GPIO01
    inverted: true
    mode:
      input: true
      pullup: true
	disable_warnings: true
  ```

However, I think blanket disabling all warnings makes it a bit too easy to shoot yourself in the foot.

closed 2023-10-19 by jesserockz

clinta, 2023-11-17:
Is this fix expected to work on output pins? I'm getting the error `[ignore_strapping_warning] is an invalid option for [output.ledc]` in esphome 2023.11.1