esphome/feature-requests #2170: (temporarily) disable device (and filter those)

HA-TB303, 2023-03-16:
**Describe the problem you have/What new integration you would like**

- The ability to temporarily disable a device which is offline.
- Disabled devices are excluded from "update all"
- Disabled devices can be visually filtered from the list (exactly like in Home Assistant device listing)

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Nosen92, 2023-10-24:
I very much welcome this feature. Some YAML configs take quite some effort to write, so the user might resist deleting them (understandably). Being able to disable and archive configs would allow you to experiment without cluttering the dashboard. It would also make it easier to return to an esphome project.

kirill-sh, 2023-11-08:
I with you. This is a much needed feature.

walzing, 2024-01-07:
Same here - I have now 10 Plug showing offline

jath58, 2024-06-12:
Same here, much needed when devices are temporarily not used (but will come back).

tevaadarelsol, 2024-08-26:
+1 for this feature request...

larsoss, 2024-11-14:
+2 ;-)

vb7445, 2024-12-20:
+3 :)

Sansei-Innovation, 2025-08-09:
Agree.  Alternatively, the ability to organize devices in (user defined nameed) folders.  This would enable the creation of "Disabled" or "Archive" folders.  Under this scenario, "_Update All_" could either be constrained to the active folder or a pop could toggle "Active Folder" or "All Folders"