esphome/feature-requests #2171: Reuse compiled source between nodes

LordMike, 2023-03-17:
**Describe the problem you have/What new integration you would like**
I've been managing a few identical devices (apart from their identity) for a little while now, and I've been following along when it's been compiling and I see the same code being compiled again and again. This makes sense of course, because it's simpler to see each device as a completely separate unit.

I imagine that in my and many other instances though, at least parts of these files will be identical. It is my impression that the device config is used to substitute values into the source (`cpp`) files, which is then compiled. This could be f.ex. the wifi configuration, which would (at minimum) contain an SSID to connect to. This SSID is the same though, on all my devices. So I'm thinking it should be possible to reuse the compiled output (the `.o` file).

If I'm right in assuming the `.cpp` files are modified with the values they need (from the device config), it should be possible to do something like:

1. Substitute in the values needed in the `.cpp` file
2. Hash the `.cpp` file, f.ex. using `md5` or something less intensive like [`xxhash`](https://github.com/Cyan4973/xxHash)
3. Check if a file exists at `/somedir/output/HASH.o`
4. If the file exists, copy it in to the output, where the output would normally go. We know this output is identical, to what we would have produced, as the source input is identical
5. If the file does not exist, compile the `.cpp`, and copy the output to `/somedir/output/HASH.o`
6. Proceed as normal, perfom linking as normal, build a firmware as normal - this should work, as all compiled output will still be in the place it usually is.

The temporary/cache location can be cleared at will - if it's empty, the sources are compiled again and placed there. Likewise, if there are access timestamps, you can remove the least recently used files in order to keep a rolling set of used files.

**Please describe your use case for this integration and alternatives you've tried:**
N/A

**Additional context**
N/A

LordMike, 2023-03-17:
I've found that the actual compilation is [handled by `platformio`](https://github.com/esphome/esphome/blob/6b7b0768755b21cca0e4b67515de93b9ca7b3821/esphome/__main__.py#L199-L207),  which means this idea either has to:

* Use features in `platformio` to share the output. But as I understand it, each esphome device is a separate environment, and as such it isn't advisable to share the build outputs.
* Before running `platformio`, do the hashing and copying of known outputs

I've also realized that the reason there are different environments is due to the different target platforms (esp82xx and whatnot). I think this is still manageable, by making the cache location dependent on these settings. If we know the list of settings that could change the build output, we can hash those together and add it to our cache directory, something like this:

```
cache_dir = "/esphome_cache/"
cache_dir += hash(settings.esp8266) + "/"
```

This way, anything changed in the `esp8266:` yaml section will produce different cache directories.

LordMike, 2023-03-17:
I've hacked the `writer.py` to set `build_cache_dir = /esphome_cache/` in the `platformio.ini`. I found that it's generated every time I build, so I couldn't just edit the file. I'll try building two identical devices, and see if it works out.

LordMike, 2023-03-18:
Update: The platformio issue has been closed with no change currently, as it probably needs a much smaller reproducible example. So I'll have to figure out how to make that eventually.

But the crux of it (https://github.com/platformio/platformio-core/issues/4574#issuecomment-1475018136) is that the underlying SCons system (which actually does the build if i understand correctly) does the hashing and determination to use the cache directory and files. It however seems to include the file path in these hashes, which means that two environments in `esphome` can never produce the same hash, as they're placed in each their own directories.

rgriffogoes, 2023-04-29:
That's a bummer! I was hoping to find a way to reduce the time needed to updated multiple devices (my ESPHome dashboard runs in a VM in my NAS, not so much compute power).

Hopefully `platformio` and/or SCons will eventually do something about it (I would imagine this improved caching mechanism would also benefit non-ESPHome cases). Thanks for the evaluation!

For now, I'm booting up a temporary ESPHome dashboard on a more powerful device, mounting same config file folders. So far so good!

petervk, 2023-12-02:
Would love something like this to be implemented as I have 18 Sonoff S31s with almost identical configurations that take a long time to update as they have to compile every single file from scratch.

LordMike, 2023-12-11:
I just saw @TMaYaD mention he had ~200 devices in [#5821](https://github.com/esphome/esphome/pull/5821). _That_ must be a pain to update :|

TMaYaD, 2023-12-11:
I have my own [wrapper around esphome](https://GitHub.com/tmayad/esphome-plus) that runs on kubernetes as a cronjob and automatically reconciles devices. All device configs are also tracked on a repo, a la. devops.

LordMike, 2023-12-19:
Oh wow - that's neat. Any chance this will make its way to the core esphome?

I've been wondering about a "version" as well, being able to determine that the local config is the same as the device's, as to avoid deploying no-ops. This seems to handle this as well :)

Expaso, 2023-12-30:
Haha Nice Wrapper @TMaYaD!
Finally some real steps to manage all the updates of all the ESPHome devices. I like it!

But 200 devices, dude! Is your Wifi still stable?

LordMike, 2024-04-23:
Just wanted to pipe in again that I made a node-red flow that updates my nodes one by one. It takes a while, but it does complete eventually. It works by listing all the `update` entities in HA of the `esphome` kind that have an update and then looping through them, waiting for their completion. Once done, it'll notify me.

[image]

<details>
  <summary>Node red flow</summary>
  
```json
[{"id":"0f02093ca01442aa","type":"comment","z":"938136b3.0c0f48","name":"Update esphome","info":"","x":400,"y":1140,"wires":[]},{"id":"476bfedb04779a9a","type":"function","z":"938136b3.0c0f48","name":"Logic","func":"const fn_updateSingleDevice = context.get('fn_updateSingleDevice');\n\n// Get all esphome nodes\nconst states = global.get('homeassistant.homeAssistant.states');\nconst filteredKeys = Object.keys(states)\n    .filter(key => /^update\\..*_firmware$/.test(key))\n    .filter(key => key in states)\n    .filter(key => {\n        let state = states[key];\n        // filter to esphome & update available\n        return state.state === \"on\" && state.attributes.title === \"ESPHome\";\n    });\n\n// For each node, emit an update\nfor (const key of filteredKeys) {\n    await fn_updateSingleDevice(key);\n}\n\nif (filteredKeys.length > 0)\n    return [null, {updated: filteredKeys}];\n\nnode.status({ fill: \"green\", shape: \"dot\", text: `updated ${filteredKeys.length} devices` });\n","outputs":2,"timeout":0,"noerr":0,"initialize":"const statesDict = global.get('homeassistant.homeAssistant.states');\n\nfunction sleep(ms = 1000) {\n    return new Promise(resolve => setTimeout(resolve, ms));\n}\n\ncontext.set('fn_updateSingleDevice', async function(key) {\n    if (!statesDict[key].attributes.in_progress)\n    {\n        node.send([{\n            entity_id: key\n        }]);\n    }\n    else {\n        node.warn(`Entity ${key} is already updating, not triggering, but waiting`);\n    }\n\n    while (statesDict[key].state !== 'off') {\n        let state = statesDict[key];\n\n        if (state.attributes.in_progress)\n            node.status({ fill: \"yellow\", shape: \"dot\", text: `updating ${key}` });\n        else\n            node.status({ fill: \"yellow\", shape: \"ring\", text: `waiting for update ${key}` });\n\n        await sleep(5000);\n    }\n\n    node.status({ fill: \"green\", shape: \"dot\", text: `updated ${key}` });\n});","finalize":"","libs":[],"x":390,"y":1240,"wires":[["192dbb924055cb60","4d89afc7d9cb0b39"],["67930ba63f8a94c2"]]},{"id":"4e3b95b38acf5c74","type":"inject","z":"938136b3.0c0f48","name":"Trigger","props":[],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":230,"y":1240,"wires":[["476bfedb04779a9a"]]},{"id":"192dbb924055cb60","type":"debug","z":"938136b3.0c0f48","name":"Trigger update","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":620,"y":1180,"wires":[]},{"id":"4d89afc7d9cb0b39","type":"api-call-service","z":"938136b3.0c0f48","name":"Trigger update","server":"af18237.61268e","version":5,"debugenabled":false,"domain":"update","service":"install","areaId":[],"deviceId":[],"entityId":["{{entity_id}}"],"data":"{}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":620,"y":1220,"wires":[[]]},{"id":"67930ba63f8a94c2","type":"api-call-service","z":"938136b3.0c0f48","name":"","server":"af18237.61268e","version":5,"debugenabled":false,"domain":"notify","service":"notify_michael","areaId":[],"deviceId":[],"entityId":[],"data":"{\t    \"message\": \"Updated \" & updated,\t    \"title\": \"Esphome updates completed\"\t}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","output_location":"","output_location_type":"none","x":640,"y":1280,"wires":[[]]},{"id":"af18237.61268e","type":"server","name":"Home Assistant","version":5,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true,"heartbeat":false,"heartbeatInterval":"30","areaSelector":"friendlyName","deviceSelector":"friendlyName","entitySelector":"friendlyName","statusSeparator":"at: ","statusYear":"hidden","statusMonth":"short","statusDay":"numeric","statusHourCycle":"h23","statusTimeFormat":"h:m","enableGlobalContextStore":true}]
```
</details>