esphome/feature-requests #2173: Support for PCF8563T RTC

fran6120, 2023-03-18:
**Describe the problem you have/What new integration you would like**

I have recently bought an RTC PCF8563T instead of PCF85063 by mistake.

Unfortunately the platform pcf85063 time integration, is not compatible with pcf8563t so the reading and writing of time and date are not correct.

**Please describe your use case for this integration and alternatives you've tried:**

The RTC PCF8563T is cheap and supports I2C communication frequencies higher than 100 kHz, it is easily found on cheap websites, so it is a good alternative to RTC DS1307

**Additional context**

KoenBreeman, 2023-04-16:
Hi Fran6120

I've tried to make the code work for the PCF8563, I found this documenation which I used. I've made a branch in development environment. I'll create a PR and will post the code to use it.

https://nl.mouser.com/ProductDetail/NXP-Semiconductors/PCF8563T-5518?qs=beN0Cyoe8YmsvsbXpQCrnw%3D%3D

Could you check if it works?

KoenBreeman, 2023-04-18:
Hi Fran6120 

I've update the time.py and removed the underscore. Also I will clean it up and hopefully the moderators will push it into the dev branch.