esphome/feature-requests #2175: Support PCF8591

tiimsvk, 2023-03-19:
**Describe the problem you have/What new integration you would like**
it is possible to add support for the PCF8591 AD/DA flood sensor.  It communicates using the i2c bus.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

whakru, 2023-05-20:
While there is no official solution made temporary for myself. You may need:
File /config/esphome/pcf8591.h

```
#include "esphome.h"
#include <Wire.h>

#define TIME_INTERVAL 1000

#define PCF8591 (0x90 >> 1)
byte value0, value1, value2, value3;

class CustomPCF8591 : public PollingComponent {
 public:

  Sensor *adc0= new Sensor();
  Sensor *adc1 = new Sensor();
  Sensor *adc2 = new Sensor();
  Sensor *adc3 = new Sensor();

  CustomPCF8591() : PollingComponent(TIME_INTERVAL) { }

  void setup() override {
    Wire.begin();
  }

  void update() override {
 Wire.beginTransmission(PCF8591); // пробуждаем PCF8591
 Wire.write(0x04); // управляющий байт - чтение ADC0, затем автоинкремент
 Wire.endTransmission(); // конец передачи
 Wire.requestFrom(PCF8591, 5);
 value0=Wire.read(); 
 value0=Wire.read(); 
 adc0->publish_state(value0);
 value1=Wire.read();
 adc1->publish_state(value1);
 value2=Wire.read(); 
 adc2->publish_state(value2);
 value3=Wire.read();
 adc3->publish_state(value3);
  }
};
```

```
esphome:
  libraries:
    - "Wire"
  includes:
    - pcf8591.h

sensor:
  - platform: custom
    lambda: |-
      auto CustomPCF = new CustomPCF8591();
      App.register_component(CustomPCF);
      return {CustomPCF->adc0 ,CustomPCF->adc1, CustomPCF->adc2, CustomPCF->adc3};
    sensors:
      - name: "My Custom adc0 Sensor"
      - name: "My Custom adc1 Sensor"
      - name: "My Custom adc2 Sensor"
      - name: "My Custom adc3 Sensor"
```

There may be inaccuracies. I did it quickly for myself .... Keep this in mind!

kroimon, 2023-05-20:
As you were already able to write a custom component, why not create an official component and send a pull request? Search the existing codebase for components implementing `i2c::I2CDevice` to find examples you could base your work on 😊

tiimsvk, 2023-05-20:
That's exactly what I was looking for.  I started something similar, but without programming experience outside esphome, I'm lost.  I will try and write the results.

whakru, 2023-05-21:
> As you were already able to write a custom component, why not create an official component and send a pull request? Search the existing codebase for components implementing `i2c::I2CDevice` to find examples you could base your work on 😊

I'm not very good at it and I don't have enough time.