esphome/feature-requests #2176: Support for ESP32-C6 chipset

linkedupbits, 2023-03-19:
**Describe the problem you have/What new integration you would like**
Provide ability to compile and flash ESPHome firmware to the new ESP32-C6 chipsets. In particular the ESP32-c6-DevKitC-1 is available now. This model contains native support for 802.14.4, and has esp-idf library support for ZigBee / Thread communication.

**Please describe your use case for this integration and alternatives you've tried:**
Build and deploy firmware to this model of chipset. This is also a first step to be able to support ZIgBee / Thread on this device (but this will be a separate process. This is just support "default" ESPHome components via native build/flash.

**Additional context**

- [Espressive's product page](https://www.espressif.com/en/products/socs/esp32-c6) for details on the ESP32-C6
- [PR to add support for the esp32-c6 in espressif's esptool-js](https://github.com/espressif/esptool-js/pull/92)

Hedda, 2023-03-20:
Copy from https://github.com/esphome/feature-requests/issues/1397 and https://github.com/esphome/feature-requests/issues/1430 for reference:

You may be interested in this fork. @stintel managed to flash a C6.

https://github.com/stintel/esphome/tree/esp32c6

**ESP32-C6** SoC, module and development board has now been launched with WiFi 6, BLE 5.0, and 802.15.4, ink. OpenThread and Zigbee!    ...so think these could be perfect for ESPHome as either a Thread Board Router (for Matter) or as a Zigbee Router device.

https://www.espressif.com.cn/en/news/ESP32-C6_Available

https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html

https://github.com/espressif/esp-idf/issues/10423

Espressif launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module:

https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/

[image]

[image]

_Espressif Systems has finally launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module available on [Aliexpress for $14.25](https://fave.co/3VYE3RW) for a pack of 5 modules as well as the ESP32-C6-DevKitC-1 development board that’s [out of stock on the official store](https://fave.co/3vU14ea) at the time of writing. but somehow listed in another store [for $16.22 plus shipping](https://s.click.aliexpress.com/e/_DlvBgFz)._

_Besides the availability announcement, another surprise is the 802.15.4 radio for Zigbee and Thread that was not part of the original [ESP32-C6 announcement](https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/) in April 2021. The ESP32-C6-DevKitC-1 development board comes with a module with 8MB flash, exposes most of the I/O’s from the ESP32-C3, and comes with two USB-C ports, two buttons, and an RGB LED._

ESP32-C6-DevKitC-1 development board specifications:

- Wireless module  – ESP32-C6-WROOM-1 with
  - ESP32-C6 single-core 32-bit RISC-V clocked up to 160 MHz 320KB ROM, 512KB SRAM, low-power RISC-V core @ up to 20 MHz
  - Wireless – 2.4 GHz WiFi 6, Bluetooth 5.0 LE, 802.15.4 radio for Zigbee 3.0 and Thread. Matter compatible. PCB antenna
8MB SPI flash
- USB – 1x USB Type-C host port, 1x USB Type-C port via UART console
- I/Os – 2x 16-pin headers with GPIO, UART, SPI, ADC, SDIO, USB, I2S, etc…
- Misc – Reset and BOOT buttons, RGB LED, J5 header/jumper for current measurement
- Power Supply (mutually exclusive options)
  - 5V via USB-UART port (recommended)
  - 5V and GND pin headers
  - 3V3 and GND pin headers
- Dimensions – 51.8 x ~20 mm

[image]

[image]

_The new ESP32-C6 module and development board work with the ESP-IDF v5.1, currently under development (see [progress report](https://github.com/espressif/esp-idf/issues/10423)), and some [basic documentation](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html) can be found on read the docs._

_If you can’t purchase an ESP32-C6 module or board just yet due to limited supplies, don’t despair as more are coming very soon as [Espressif tells](https://www.espressif.com.cn/en/news/ESP32-C6_Available?position=0&list=WQQQ-4A66v8oesBoFKp-rNz9bSQ9dg5w7wc_nNvDt_I) us to expect more products on Adafruit, Akizuki, Digikey, and Mouser._

stintel, 2023-03-20:
The ESP32-C6 requires ESP-IDF 5.1, which isn't released yet. However, I was able to build a working esphome firmware for the C6 using ESP-IDF master, and esphome dev +  #4364 and #4377. Meanwhile I cleaned up #4364 and I believe it is ready for merging. Would be nice if some people could review or test that and give feedback.

Hedda, 2023-03-20:
Any chance you would also be willing to look at adding support for ESP32-H2 (ESP32-H2-MINI-1 / ESP32-H2-DevKitM-1) as well?

At least on paper, the ESP32-H2 looks to need the same config ESP32-C6 but the ESP32-H2 SoC is without Wi-Fi radio support:

https://www.espressif.com/en/products/socs/esp32-h2

https://www.espressif.com/en/news/ESP32_H2

https://media.digikey.com/pdf/Data%20Sheets/Espressif%20PDFs/ESP32-H2_MINI_1_Prelim_v0.2_EN.pdf

https://github.com/espressif/esp-idf/search?q=esp32h2

stintel, 2023-03-20:
> Any chance you would also be willing to look at adding support for ESP32-H2 (ESP32-H2-MINI-1 / ESP32-H2-DevKitM-1) as well?

If that was directed at me: I don't own a H2, and due to lack of Wi-Fi radio on the H2 and lack of Thread/Zigbee support in ESPHome, I have no use for it either.

linkedupbits, 2023-03-20:
> The ESP32-C6 requires ESP-IDF 5.1, which isn't released yet. 
I have just set up a new esp-idf dev environment using VSCode (on windows) and just added the esp-idf extension (v1.6) but this seems to have pulled in esp-idf v5.1 (based on file names in the ~/.espressif folder - is there an canonical way to tell?) Maybe 5.1 is released? (I have built and installed the sample Zigbee projects for the ESP32-C6 using vscode)

> If you use my branch, make sure to read the commit messages. I'll try to clean up the instructions in [stintel/esphome@fbb2dfe](https://github.com/stintel/esphome/commit/fbb2dfe26d74e4dd49a25030b280dee7da64f386) after work, as some things are no longer needed.

Should we fork from your branch and submit PRs back to you if we fix anything?

Thanks for your work.

stintel, 2023-03-20:
> @stintel - Should we fork from your branch and submit PRs back to you if we fix anything?

As I have not yet created a PR here, feel free to just pull from my branch, and open the PR here yourself.

Hedda, 2023-04-05:
FYI, [01Space](https://github.com/01Space) is sell a tiny ESP32-C6 IoT dev board based on ESP32-C6-WROOM-1 wireless module with two 10-pin headers + USB-C for power:

https://www.aliexpress.us/item/3256805225446433.html

[image]

Again, ESP32-C6 Arduino support will require ESP IDF 5.1 framework which is still under development:

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/issues/7852

lboue, 2023-05-25:
@Hedda 
@stintel 
@linkedupbits 

Espressif released ESP-IDF Pre-release v5.1-beta1 Pre-release recently.
So I think we can start doing some testing. t to add C6 card support in the beta branch of ESPHome. What do you think?

[ESP-IDF doc for  ESP32-C6 pre-release v5.1-beta1](https://docs.espressif.com/projects/esp-idf/en/v5.1-beta1/esp32c6/get-started/index.html)

platini76, 2023-06-13:
I am also interested.. I have one..but not recognized with esphome or web installation.. how can I do?

lboue, 2023-06-13:
> The ESP32-C6 requires ESP-IDF 5.1, which isn't released yet. However, I was able to build a working esphome firmware for the C6 using ESP-IDF master, and esphome dev + [esphome/esphome#4364](https://github.com/esphome/esphome/pull/4364) and [esphome/esphome#4377](https://github.com/esphome/esphome/pull/4377). Meanwhile I cleaned up [esphome/esphome#4364](https://github.com/esphome/esphome/pull/4364) and I believe it is ready for merging. Would be nice if some people could review or test that and give feedback. Same goes for [platformio/platform-espressif32#1054](https://github.com/platformio/platform-espressif32/pull/1054). These things needs to happen first, without them supporting ESP32-C6 isn't possible.

@stintel Did you try ESP-IDF Pre-release v5.1-beta1?

platini76, 2023-06-13:
@lboue do you expect it will be possible soon? or late?

stintel, 2023-06-14:
> @stintel Did you try ESP-IDF Pre-release v5.1-beta1?

I didn't. Other priorities right now.

stintel, 2023-06-20:
Unfortunately it's not possible to specify 5.1-rc1 in the YAML file like below, so using 5.1-rc1 still requires the same hack as documented in stintel/esphome@f369b9e36d4a5955dc0670c3ca94ddb43b9e0558:
```YAML
esp32:
  board: esp32-c6-devkitc-1
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1-rc1
```

It does seem the platformio registry now has GCC 12.2.0, so that hack is no longer necessary. We can just point to do this in:

```YAML
    esphome:
      name: "esp32c6-test2"
      platformio_options:
        platform_packages:
          - espressif/toolchain-riscv32-esp @ 12.2.0+20230208
```

My esp32c6 C6 branch is rebased on current dev. Unfortunately the other issues in the commit message are still there.

Hedda, 2023-06-21:
Based on Espressif's previous release history believe assume it should not be very long before ESP-IDF v5.1 GA release is released:

https://github.com/espressif/esp-idf/tags

https://github.com/espressif/esp-idf/releases

https://github.com/espressif/arduino-esp32/issues/7852

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/pull/7733

lboue, 2023-07-01:
@Hedda 
[ESP-IDF Release v5.1](https://github.com/espressif/esp-idf/releases/tag/v5.1) has been released.

juanimoni, 2023-07-12:
Hi, does this mean the ESP32-c6 is supported now ?

Hedda, 2023-07-14:
> does this mean the ESP32-c6 is supported now ?

No, it just means that ESP32-C6 is supported in the upsteam ESP-IDF SDK/libraries which ESPHome is dependent on. 

Instead keep an eye on https://github.com/esphome/esphome/pull/4377 which should bring initial ESP32-C6 support in ESPHome (which does not include Zigbee or Thread support).

stintel, 2023-07-26:
I have just pushed a change to https://github.com/stintel/platform-espressif32/tree/esp32-c6-test that fixes the linker error I've been hunting for months when using ESP-IDF 5.1.

I've also updated https://github.com/stintel/esphome/tree/esp32c6 and the commit message. With the info in there it should be fairly easy to build ESPHome for the ESP32-C6. Does anyone know who to poke to get ESP-IDF 5.1 added to the PlatformIO registry?

stintel, 2023-08-03:
ESP-IDF 5.1 was published on the PlatformIO registry: https://registry.platformio.org/tools/platformio/framework-espidf/versions

Will update https://github.com/stintel/platform-espressif32/tree/esp32-c6-test and https://github.com/stintel/esphome/tree/esp32c6.

udarntors, 2023-08-04:
>

With the risk of sounding stupid, I would really like to try this build on my Olimex esp32c6-evb boards, but I can't make it compile.
Actually, it does not get pass this:
```sh
$ esphome compile test.yaml
INFO ESPHome 2023.8.0-dev
INFO Reading configuration test.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

esp32: [source test.yaml:6]

  This board is unknown, please set the variant manually.
  board: esp32-c6-devkitc-1
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0
```
Thanks anyway for your time and effort!!

eloebl, 2023-08-04:
> Olimex esp32c6-evb

I also have this board and very much hoping to get it to build as well.

stintel, 2023-08-04:
You need to run ESPHome built from my [esp32c6 branch](https://github.com/stintel/esphome/tree/esp32c6).

FutureCow, 2023-08-04:
@stintel will you also create a PR so support voor the c6 will be in a future release of esphome?

stintel, 2023-08-04:
I will. I've already created esphome/esphome#5176 to prevent duplicate MAC addresses on ESP32-C6 and ESP32-H2. Will create a few more small PRs to fix some incompatibilities with ESP-IDF 5.1, cleanup my branch a bit, and finally create a PR for ESP32-C6 support.

udarntors, 2023-08-04:
Once I learned how to change git branches, it worked nicely. 
For the first boot up it crashed in a loop, I needed to add the flash size: 
```yaml
esphome:
  name: "esp32c6-170"
  platformio_options:
    board_upload.flash_size: 4MB
```
I tried to compile some switch options, but it gives an error and does not compile.

```
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp32c6-170/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::switch_command(esphome::api::SwitchCommandRequest const&)':
/Users/radu/wk/.esphome/build/esp32c6-170/src/esphome/components/api/api_connection.cpp:489: undefined reference to `esphome::switch_::Switch::turn_on()'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/radu/wk/.esphome/build/esp32c6-170/src/esphome/components/api/api_connection.cpp:489: undefined reference to `esphome::switch_::Switch::turn_off()'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp32c6-170/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/Users/radu/wk/.esphome/build/esp32c6-170/src/esphome/core/controller.cpp:35: undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp32c6-170/src/main.o: in function `setup()':
/Users/radu/wk/.esphome/build/esp32c6-170/src/main.cpp:123: undefined reference to `esphome::switch_::Switch::Switch()'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp32c6-170/src/main.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/Users/radu/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/basic_string.h:795: undefined reference to `vtable for esphome::gpio::GPIOSwitch'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/radu/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/12.2.0/bits/basic_string.h:795: undefined reference to `vtable for esphome::gpio::GPIOSwitch'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/esp32c6-170/src/main.o: in function `setup()':
/Users/radu/wk/.esphome/build/esp32c6-170/src/main.cpp:124: undefined reference to `vtable for esphome::gpio::GPIOSwitch'
/Users/radu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/radu/wk/.esphome/build/esp32c6-170/src/main.cpp:124: undefined reference to `vtable for esphome::gpio::GPIOSwitch'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32c6-170/firmware.elf] Error 1
======================================================= [FAILED] Took 15.41 seconds =======================================================
``` 
In any case, it gives hope. 👍

stintel, 2023-08-04:
@udarntors click the 3 dots, Clean Build Files, and retry.

udarntors, 2023-08-05:
> @udarntors click the 3 dots, Clean Build Files, and retry.

Thank you !!! 
After cleaning the build files, it did compile. 

I do not know if you mentioned it, but OTA works like a charm!

I have however another problem, the boards have 4 optocouplers, trying to configure them as binary sensors fails.

```
In file included from src/esphome/components/binary_sensor/automation.cpp:2:
src/esphome/components/binary_sensor/automation.cpp: In member function 'void esphome::binary_sensor::MultiClickTrigger::on_state_(bool)':
src/esphome/components/binary_sensor/automation.cpp:25:21: error: format '%u' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |                     ^~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:69:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   69 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:151:28: note: in expansion of macro 'esph_log_v'
  151 | #define ESP_LOGV(tag, ...) esph_log_v(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:25:7: note: in expansion of macro 'ESP_LOGV'
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |       ^~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:25:33: note: format string is defined here
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |                                ~^
      |                                 |
      |                                 unsigned int
      |                                %lu
src/esphome/components/binary_sensor/automation.cpp:25:21: error: format '%u' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |                     ^~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:69:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   69 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:151:28: note: in expansion of macro 'esph_log_v'
  151 | #define ESP_LOGV(tag, ...) esph_log_v(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:25:7: note: in expansion of macro 'ESP_LOGV'
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |       ^~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:25:40: note: format string is defined here
   25 |       ESP_LOGV(TAG, "START min=%u max=%u", evt.min_length, evt.max_length);
      |                                       ~^
      |                                        |
      |                                        unsigned int
      |                                       %lu
src/esphome/components/binary_sensor/automation.cpp:54:19: error: format '%u' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   54 |     ESP_LOGV(TAG, "A i=%u min=%u max=%u", *this->at_index_, evt.min_length, evt.max_length);  // NOLINT
      |                   ^~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:69:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
   69 | #define ESPHOME_LOG_FORMAT(format) format
      |                                    ^~~~~~
src/esphome/core/log.h:151:28: note: in expansion of macro 'esph_log_v'
  151 | #define ESP_LOGV(tag, ...) esph_log_v(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:54:5: note: in expansion of macro 'ESP_LOGV'
   54 |     ESP_LOGV(TAG, "A i=%u min=%u max=%u", *this->at_index_, evt.min_length, evt.max_length);  // NOLINT
      |     ^~~~~~~~
src/esphome/components/binary_sensor/automation.cpp:54:32: note: format string is defined here
   54 |     ESP_LOGV(TAG, "A i=%u min=%u max=%u", *this->at_index_, evt.min_length, evt.max_length);  // NOLINT
      |                               ~^
      |                                |
      |                                unsigned int
      |                               %lu
src/esphome/components/binary_sensor/automation.cpp:54:19: error: format '%u' expects argument of type 'unsigned int', but argument 7 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   54 |     ESP_LOGV(TAG, "A i=%u min=%u max=%u", *this->at_index_, evt.min_length, evt.max_length);  // NOLINT
      |                   ^~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:69:36: note: in definition of macro 'ESPHOME_LOG_FORMAT'
[... 64 lines cut ...]
   71 |   ESP_LOGV(TAG, "Multi Click: Invalid length of press, starting cooldown of %u ms...", this->invalid_cooldown_);
      |                                                                             ~^
      |                                                                              |
      |                                                                              unsigned int
      |                                                                             %lu
Compiling .pioenvs/esp32c6-170/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.o
[... cut ...]
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esp32c6-170/src/esphome/components/gpio/switch/gpio_switch.o
Compiling .pioenvs/esp32c6-170/src/esphome/components/logger/logger.o
cc1plus: some warnings being treated as errors
*** [.pioenvs/esp32c6-170/src/esphome/components/binary_sensor/automation.o] Error 1
======================================================= [FAILED] Took 17.86 seconds =======================================================
``` 
I did try the `clean build files` trick, does not work on this one.

stintel, 2023-08-05:
Try this:
```
esphome:
  name: "esp32c6-170"
  platformio_options:
    board_upload.flash_size: 4MB
    build_flags:
      - -Wno-format
```

udarntors, 2023-08-05:
That worked!! This is great, I want to use the boards for controlling lights around the house, the optocouplers give the ability to use the already mounted physical switches and just toggle the relays.
Here is the working config.yaml
```yaml
esphome:
  name: "esp32c6-171"
  platformio_options:
    board_upload.flash_size: 4MB
    build_flags:
      - -Wno-format

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0

logger:
  level: VERBOSE
# VERY_VERBOSE not working
api:
  encryption:
    key: "cTfIIkUnjEEeL/APTJIQVpsyD0ff+IRIjxXqclDnJAqU="

ota:
  password: "1456dogfood564"

wifi:
  ssid: nova
  password: 182526182526a
  power_save_mode: LIGHT
  fast_connect: true
  manual_ip:
    static_ip: 192.168.0.171
    gateway: 192.168.0.254
    subnet: 255.255.255.0
    dns1: 192.168.0.202

switch:
  - platform: gpio
    id: rel_1
    name: "Relay 1"
    pin: 10
  - platform: gpio
    name: "Relay 2"
    id: rel_2
    pin: 11
  - platform: gpio
    name: "Relay 3"
    id: rel_3
    pin: 22
  - platform: gpio
    name: "Relay 4"
    id: rel_4
    pin: 23

status_led:
  pin:
    number: 8
    inverted: True

binary_sensor:
  #- platform: status
  #  name: status
  - platform: gpio
    name: opto 1
    pin:
      number: 1
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      then:
        - switch.toggle: rel_1

  - platform: gpio
    name: opto 2
    pin:
      number: 2
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      then:
        - switch.toggle: rel_2
  - platform: gpio
    name: opto 3
    pin:
      number: 3
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      then:
        - switch.toggle: rel_3

  - platform: gpio
    name: opto 4
    pin:
      number: 15
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      then:
        - switch.toggle: rel_4

```

Thanks again!!

ducky64, 2023-08-06:
Was also able to get this working on a custom board, with ESP32C6 module from Mouser
- Working: 
  - LED GPIO
  - Switch (button) GPIO
  - BH1750 illuminance sensor over I2C
  - WiFi, HASS API, webserver
- Not working:
  - RTTTL (the ESP device crashes and reboots)
  - SNTP (fails to compile)
  - ADC (maybe? stack traces when passed in a bad pin, without a helpful error message)
  - Anything requiring Arduino (which I guess is expected at this stage)

enrique, 2023-08-08:
I've tried ESPhome in home Assistant, but can't get it to work.
I am using the "stintel:esp32c6" branch in the configuration, and when I check my yaml, it says OK. But when installing it gives me the following message: "Configuration does not match the platform of the connected device. Expected an ESP32C3 device."
```yaml
---
esphome:
  # name: c6
  name: "esp32c6-test"

  friendly_name: C6
  platformio_options:
    board_upload.flash_size: 8MB
#    platform_packages:
#    - espressif/toolchain-riscv32-esp @ 12.2.0+20230208
#    build_flags: -Wno-format

esp32:
  board: "esp32-c6-devkitc-1"
  variant: "ESP32C6"
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    # source: github://stintel/esphome@esp32c6
    # source: github://stintel/esphome/tree/esp32c6
    # source: github://stintel/platform-espressif32/esp32-c6-test
    #                https://github.com/stintel/platform-espressif32/tree/esp32-c6-test
    type: esp-idf
    version: 5.1.0

# external_components:
  # - source: github://pr#
  #source: github://stintel/esphome@esp32c6
  # type: git
  # source: github://stintel/esphome/tree/esp32c6
  #  components:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "C6 Fallback Hotspot"
    password: !secret esphome_fallback_ap_password

captive_portal:

```
What am I missing?

rotilho, 2023-08-08:
You need to run the esphome from his branch as well.

stintel, 2023-08-08:
> What am I missing?

Someone on Discord had the same problem. Flashing via web tools doesn't work. Download the binary and flash via esptool directly, or passthrough the USB device to the container.

leoshusar, 2023-08-08:
Can anyone please confirm if UART already works? (I see logger component here I suppose it does, but just to be sure :) )
I'm asking because I'm ordering parts for my new project which uses RS232 and I'm deciding if I should already get a C6.

udarntors, 2023-08-08:
> What am I missing?
Be sure to have only the esphome dev variant installed on your hass.

ducky64, 2023-08-08:
@leoshusar UART works, at least with the logging component

udarntors, 2023-08-09:
> @udarntors 🤦‍♂️ Sure, I have both installed. Can it be running one of them each time or do I have to uninstall it?

I think the error is caused by the fact you have both installed, so uninstall the standard and leave only the dev one with `stintel:esp32c6` branch configured.

enrique, 2023-08-09:
Iv'e got the same mesage on every try until...
- Uninstaled both ESPHome (dev and standard) 
- Reinstaled dev version only.
- Configured ESPHome (dev) with `stintel:esp32c6` banch.
- Once running, selected, from the device three dots menu, install/manual download/modern format.
- Upload bin through web.esphome.io install (need to press boot button)

After this, even OTA works as expected.

My yaml config:
```yaml
---
esphome:
  name: "esp32c6"
  friendly_name: C6
  platformio_options:
    board_upload.flash_size: 8MB

esp32:
  board: "esp32-c6-devkitc-1"
  variant: "ESP32C6"
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "C6 Fallback Hotspot"
    password: !secret esphome_fallback_ap_password

web_server:
  port: 80
  auth:
    username: admin
    password: !secret web_server_password

captive_portal:
``` 

Log after upload bin 🥇 
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x17e8
entry 0x4086c410
I (23) boot: ESP-IDF 5.0.2 2nd stage bootloader
I (24) boot: compile time Aug  9 2023 15:29:21
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 40MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (72) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (79) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (87) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (94) boot: End of partition table
I (99) boot: No factory image, trying OTA 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=40800000 size=1078ch ( 67468) load
I (128) esp_image: segment 1: paddr=000207b4 vaddr=40810790 size=030f8h ( 12536) load
I (131) esp_image: segment 2: paddr=000238b4 vaddr=00000000 size=14764h ( 83812) 
I (150) esp_image: segment 3: paddr=00038020 vaddr=420a8020 size=2f568h (193896) map
I (190) esp_image: segment 4: paddr=00067590 vaddr=00000000 size=08a88h ( 35464) 
I (198) esp_image: segment 5: paddr=00070020 vaddr=42000020 size=a3578h (669048) map
I (339) boot: Loaded app from partition at offset 0x10000
I (366) boot: Set actual ota_seq=1 in otadata[0]
I (366) boot: Disabling RNG early entropy source...
I (377) cpu_start: Unicore app
I (377) cpu_start: Pro cpu up.
W (386) clk: esp_perip_clk_init() has not been implemented yet
I (392) cpu_start: Pro cpu start user code
I (392) cpu_start: cpu freq: 160000000 Hz
I (393) cpu_start: Application information:
I (395) cpu_start: Project name:     esp32c6
I (400) cpu_start: App version:      2023.8.0-dev
I (406) cpu_start: Compile time:     Aug  9 2023 15:28:52
I (412) cpu_start: ELF file SHA256:  0000000000000000...
I (418) cpu_start: ESP-IDF:          5.0.2
I (422) cpu_start: Min chip rev:     v0.0
I (427) cpu_start: Max chip rev:     v0.99 
I (432) cpu_start: Chip rev:         v0.0
I (437) heap_init: Initializing. RAM available for dynamic allocation:
I (444) heap_init: At 40818700 len 00063F10 (399 KiB): D/IRAM
I (450) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (457) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (464) spi_flash: detected chip: generic
I (468) spi_flash: flash io: dio
I (472) sleep: Configure to isolate all GPIO pins in sleep state
I (479) sleep: Enable automatic switching of GPIO sleep configuration
I (486) coexist: coex firmware version: ebddf30
I (491) coexist: coexist rom version 5b8dcfa
I (496) app_start: Starting scheduler on CPU0
I (501) main_task: Started on CPU0
I (505) main_task: Calling app_main()
```

Hedda, 2023-08-21:
FYI, [vidplace7](https://github.com/vidplace7)'s & [stintel](https://github.com/stintel)'s pull request with support for ESP32-C2 & ESP32-C6 has now been merged -> https://github.com/esphome/esphome/pull/4377

PS: Note that it does not include Zigbee or Thread (or Matter) support so if wishing that then see this feature request -> https://github.com/esphome/feature-requests/issues/1397

tortfeaser, 2023-08-30:
Any ideas why the following YAML won't compile? This is on a ESP32C6 with 8MB flash, marked nanoESP32-C6 v1.0 with chip_id: 
`esptool.py v4.6.2
Serial port COM9
Connecting....
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:41:f5:74
BASE MAC: 40:4c:ca:41:f5:74
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C6 has no Chip ID. Reading MAC instead.
MAC: 40:4c:ca:ff:fe:41:f5:74
BASE MAC: 40:4c:ca:41:f5:74
MAC_EXT: ff:fe
Hard resetting via RTS pin...` 
and flash_id: `Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:41:f5:74
BASE MAC: 40:4c:ca:41:f5:74
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB`

YAML:
```esphome:
  name: "esp32c6"
  platformio_options:
    board_upload.flash_size: 8MB

esp32:
  board: esp32-c6-devkitc-1
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0
    
substitutions:
  device_name: "esp32c6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret wifi_password

captive_portal:
    
# Enable debug component
debug:

# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE
  logs:
    sensor: INFO
    duty_cycle: INFO
    binary_sensor: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret esphome_key
  reboot_timeout: 0s

ota:
  password: !secret default_password

# used just for testing, otherwise no need for it
web_server:
  port: 80

time:
  - platform: homeassistant
  ```
Build log is attached. File won't compile with `build_flags: -Wno-format` set either.

[logs_esp32c6_compile.txt](https://github.com/esphome/feature-requests/files/12471762/logs_esp32c6_compile.txt) or at https://pastebin.com/pz3E5XzF

stintel, 2023-08-30:
I'd try disabling the debug component for now. Can have a deeper look next month.

stintel, 2023-09-07:
This feature request can be closed since fe7893d1b3178912ad8e016d0ffba2bf6836928d. Problems should be reported as issues.

closed 2023-09-07 by kbx81

stintel, 2023-09-07:
@tortfeaser esphome/esphome#5354

tortfeaser, 2023-09-07:
Thanks for this. I installed the dev branch in Home Assistant, and tried to upload to an ESP32C6 8MB

Chip is 
```
esptool.py v4.6.2
Serial port COM9
Connecting....
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
```

When trying to install using the HA interface via USB/serial, I get 
[image]

When compiling to try and upload using esptool or the ESP-Home flasher, I get this bunch of errors:
[logs_esp32c6_compile (2).txt](https://github.com/esphome/feature-requests/files/12554093/logs_esp32c6_compile.2.txt)

The YAML I'm trying to compile is:
```
esphome:
  name: "esp32c6"

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.1
    
substitutions:
  device_name: "esp32c6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret wifi_password

captive_portal:
    
# Enable debug component
debug:

# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE
  logs:
    sensor: INFO
    duty_cycle: INFO
    binary_sensor: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret esphome_key
  reboot_timeout: 0s

ota:
  password: !secret default_password

# used just for testing, otherwise no need for it
web_server:
  port: 80

time:
  - platform: homeassistant
```

digiblur, 2023-09-11:
Nice!  I was able to get an esp32-c6 to compile but it looks like it won't compile with BT proxy configs yet.  

```In file included from src/esphome/components/esp32_ble/ble.cpp:7:
src/esphome/components/esp32_ble/ble.cpp: In member function 'bool esphome::esp32_ble::ESP32BLE::ble_setup_()':
/config/.esphome/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]```

ThomPre99, 2023-09-13:
Unfortunately the compile process always aborts with an error message:

```
INFO ESPHome 2023.8.3
INFO Reading configuration /config/esphome/esp32-c6-1.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

esp32: [source /config/esphome/esp32-c6-1.yaml:9]
  board: esp32-c6-devkitc-1
  
  Unknown value 'ESP32C6', did you mean 'ESP32C3', 'ESP32', 'ESP32S3'?.
  variant: ESP32C6
  framework: 
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0
```

The yaml-file is:

```
esphome:
  name: "esp32-c6-1"
  platformio_options:
    board_upload.flash_size: 4MB
    build_flags:
      - -Wno-format

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
    type: esp-idf
    version: 5.1.0

logger:
  level: VERBOSE
# VERY_VERBOSE not working
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: LIGHT
  fast_connect: true
  manual_ip:
    static_ip: 192.168.178.xx
    gateway: 192.168.178.xx
    subnet: 255.255.255.xx
    dns1: 192.168.178.xx
```

The entry "variant: ESP32C6" is underlined in red.

What am I doing wrong?

stintel, 2023-09-13:
> What am I doing wrong?

You need to run esphome from the dev branch, or wait for the 2023.09 release.

mmib, 2023-09-17:
> Nice! I was able to get an esp32-c6 to compile but it looks like it won't compile with BT proxy configs yet.
> 
> ```
> src/esphome/components/esp32_ble/ble.cpp: In member function 'bool esphome::esp32_ble::ESP32BLE::ble_setup_()':
> /config/.esphome/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]```
> ```

got same problem. Was able to compile and run basic ESPHOME things but not bluetooth_proxy

digiblur, 2023-09-18:
> > Nice! I was able to get an esp32-c6 to compile but it looks like it won't compile with BT proxy configs yet.
> > ```
> > src/esphome/components/esp32_ble/ble.cpp: In member function 'bool esphome::esp32_ble::ESP32BLE::ble_setup_()':
> > /config/.esphome/platformio/packages/framework-espidf/components/bt/include/esp32c6/include/esp_bt.h:263:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]```
> > ```
> 
> got same problem. Was able to compile and run basic ESPHOME things but not esp32_ble_tracker and bluetooth_proxy

try     version: 5.1.0  in your yaml

mmib, 2023-09-18:
another error with 5.1.0
`Linking .pioenvs/co2/firmware.elf
RAM:   [=         ]   6.5% (used 34264 bytes from 524288 bytes)
Flash: [======    ]  55.6% (used 1020826 bytes from 1835008 bytes)
Building .pioenvs/co2/firmware.bin
Creating esp32c6 image...

A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
*** [.pioenvs/co2/firmware.bin] Error 2
`

mmib, 2023-09-18:
ok, seems working with platform_version: https://github.com/stintel/platform-espressif32#esp32-c6-test
not working with https://github.com/platformio/platform-espressif32 (but it was fine with 5.1.1 without BLE)

mmib, 2023-09-18:
I was able to compile but Wi-Fi is not working together with bluetooth_proxy while working fine if compile with 5.1.0 but without bluetooth_proxy
```

[22:41:56][I][wifi:277]: WiFi Connecting to 'network_name'...
[22:41:56][D][esp-idf:000]: W (45168) esp_adapter: wifi_reset_mac_wrapper() has not been implemented yet
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45168) wifi:
[22:41:56][D][esp-idf:000]: (bf)761:0x600a7cac:0x01b4b4b0
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45172) wifi:
[22:41:56][D][esp-idf:000]: (agc)0x600a7128:0xd2002800, min.avgNF:0xce->0xd2(dB), RCalCount:0x2, min.RRssi:0x800(-128.00)
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45182) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS0:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45203) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS1:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45213) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS2:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45223) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS3:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45233) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS4:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45244) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS5:19
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45254) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS6:18
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45264) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS7:18
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45274) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS8:17
[22:41:56][D][esp-idf:000]: 
[22:41:56]
[22:41:56][D][esp-idf:000]: W (45285) wifi:
[22:41:56][D][esp-idf:000]: (TB)WDEV_PWR_TB_MCS9:15
[22:41:56][D][esp-idf:000]: 
[22:41:56]

```

platini76, 2023-09-28:
any guide to use it directly with espmhome? ready

platini76, 2023-09-28:
> * stintel:esp32c6

my config:

home_assistant_dashboard_integration: false
esphome_fork: stintel:esp32c6

but I got an error:
[01:52:04] INFO: Checking forked ESPHome
[01:52:04] INFO: Downloading ESPHome from fork 'pull/stintel/head' (https://github.com/esphome/esphome/archive/pull/stintel/head.tar.gz)...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0

100    14  100    14    0     0     26      0 --:--:-- --:--:-- --:--:--    26
[01:52:04] INFO: Installing ESPHome from fork 'pull/stintel/head' (https://github.com/esphome/esphome/archive/pull/stintel/head.tar.gz)...
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
[01:52:04] FATAL: Failed installing ESPHome from fork.
cont-init: info: /etc/cont-init.d/30-esphome-fork.sh exited 1
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.

stintel, 2023-09-29:
ESPHome 2023.9 was released yesterday. It has support for the ESP32-C6. No need to run my fork, as mentioned earlier in this issue. There is also working YAML code to be found.

platini76, 2023-09-29:
[image]
  with 2023.9 I got this errot

stintel, 2023-09-29:
See https://github.com/esphome/esphome/pull/5289. If you still can't figure it out I suggest you try Discord.

leoshusar, 2023-09-29:
I finally tried the UART component and I'm running into this error:
```
src/esphome/components/uart/uart_component_esp_idf.cpp: In member function 'uart_config_t esphome::uart::IDFUARTComponent::get_config_()':
src/esphome/components/uart/uart_component_esp_idf.cpp:51:28: error: 'UART_SCLK_APB' was not declared in this scope; did you mean 'UART_SCLK_RTC'?
   51 |   uart_config.source_clk = UART_SCLK_APB;
      |                            ^~~~~~~~~~~~~
      |                            UART_SCLK_RTC
```

I found [this](https://github.com/espressif/esp-idf/issues/12314) issue with the same error in the comment below.
Is this something IDF needs to get fixed or, since it's in a `esphome/components/uart/uart_component_esp_idf.cpp` file, should I open issue here?

amzaldua, 2023-11-02:
```
esphome:
  name: esp32-c6
  friendly_name: ESP32-C6

esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    type: esp-idf
```
[image]

ThomPre99, 2023-11-07:
```
esp32:
  board: esp32-c6-devkitc-1
  variant: ESP32C6
  framework:
    platform_version: https://github.com/platformio/platform-espressif32
    type: esp-idf
    version: 5.1.1
```

This configuration works for me.

[57 more replies]