esphome/feature-requests #2180: Error in Dockerfile: /usr/bin/env: ‘python3\r’: No such file or directory

Deny11Matrix, 2023-03-22:
Hi, 
i am trying to exec the dockerfile under windows and i have this error message:

#11 35.77 /usr/bin/env: ‘python3\r’: No such file or directory
------
executor failed running [/bin/sh -c pip3 install --no-cache-dir -r /requirements.txt -r /requirements_optional.txt    && /platformio_install_deps.py /platformio.ini]: exit code: 127

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

RoboMagus, 2023-03-29:
Sounds like an *issue*, not a *feature request*. You should probably move this over to [esphome/issues](https://github.com/esphome/issues/issues)

ssieb, 2023-03-29:
You need to provide more information since a lot of people use the docker method and no one else has mentioned this.  I was going to transfer this, but then I realized that creating a new issue would be better because it asks you for the information you need to provide.

closed 2023-03-29 by ssieb

randybb, 2023-03-29:
Since when is docker supported on windows?