esphome/feature-requests #2181: Add 4-channel relay i2C from seeed

cocoffr, 2023-03-23:
Can you add the board 4-channel relay i2 from seeed ? see : https://wiki.seeedstudio.com/Grove-4-Channel_SPDT_Relay/

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

KoenBreeman, 2023-04-14:
@cocoffr Can you check my code. Since I have no Seeed relay I cannot test this new feature. I hope it works

cocoffr, 2023-04-14:
hi, i seem it don't work.
my esp32.yaml :
`esphome:
  name: esp32
  friendly_name: esp32

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "/J18gQ3V4lyN1uLQDpy0EIvBxgq+zD/VSd0uYVLVUFs="

ota:
  password: "e444af504fa528d0922e80382b5102bc"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  sda: 1
  scl: 0
  scan: true
  id: bus_a

external_components:
  - source: github://pr#4684
    components: seeedmultichannelrelay

seeedmultichannelrelay:
  id: SeedMultiChannelRelay_ID
  address: 0x11 # changes the address of the seeedmultichannelrelay (only runs once to update into the EPROM) then you have to change the i2c_address

switch:
  - platform: seeedmultichannelrelay
    id: relay_1
    name: relay1
    channel: relay_1
    seeedmultichannelrelay_id: SeedMultiChannelRelay_ID`

my log:
`INFO Reading configuration /config/esphome/esp32.yaml...
ERROR Unable to import component seeedmultichannelrelay.switch:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/9965e14f/esphome/components/seeedmultichannelrelay/switch/__init__.py", line 4, in <module>
    from esphome.const import CONF_CHANNEL, CONF_INTERLOCK, CONF_INTERLOCK_WAIT_TIME
ImportError: cannot import name 'CONF_INTERLOCK_WAIT_TIME' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

switch.seeedmultichannelrelay: [source /config/esphome/esp32.yaml:40]
  
  Platform not found: 'switch.seeedmultichannelrelay'.
  platform: seeedmultichannelrelay
  id: relay_1
  name: relay1
  channel: relay_1
  seeedmultichannelrelay_id: SeedMultiChannelRelay_ID`

KoenBreeman, 2023-04-14:
Hi Cocoffr

I also had that problem during my testing here. The tests require that the variable CONF_INTERLOCK_WAIT_TIME has to be added to the const.py. For now I've update the init.py, so you can perform the tests. It should not give this error anymore. Maybe you have to give it a "clean build files". Please let me know if this version works.

KoenBreeman, 2023-04-14:
Also be aware that with this line you change the i2c address of your relay
_address: 0x11 # changes the address of the seeedmultichannelrelay (only runs once to update into the EPROM) then you have to change the i2c_address_

So if you just want the standard i2c address remove this line.

cocoffr, 2023-04-14:
I download "clean build files" but there is the same error
```
INFO Reading configuration /config/esphome/esp32-5v.yaml...
ERROR Unable to import component seeedmultichannelrelay.switch:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/9965e14f/esphome/components/seeedmultichannelrelay/switch/__init__.py", line 4, in <module>
    from esphome.const import CONF_CHANNEL, CONF_INTERLOCK, CONF_INTERLOCK_WAIT_TIME
ImportError: cannot import name 'CONF_INTERLOCK_WAIT_TIME' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

switch.seeedmultichannelrelay: [source /config/esphome/esp32-5v.yaml:45]
  
  Platform not found: 'switch.seeedmultichannelrelay'.
  platform: seeedmultichannelrelay
  id: relay_1
  name: relay1
  channel: relay_1
  seeedmultichannelrelay_id: SeedMultiChannelRelay_ID
```

KoenBreeman, 2023-04-14:
Can you remove the folders in your ESPHome\config\.esphome\external_components folder. There are all the custom components located. This way it will download the github repository again.

cocoffr, 2023-04-14:
I do what you want (delete all the folder and load new configuration) but i have the same error
```
INFO Reading configuration /config/esphome/esp32-5v.yaml...
ERROR Unable to import component seeedmultichannelrelay.switch:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/9965e14f/esphome/components/seeedmultichannelrelay/switch/__init__.py", line 4, in <module>
    from esphome.const import CONF_CHANNEL, CONF_INTERLOCK, CONF_INTERLOCK_WAIT_TIME
ImportError: cannot import name 'CONF_INTERLOCK_WAIT_TIME' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

switch.seeedmultichannelrelay: [source /config/esphome/esp32-5v.yaml:45]
  
  Platform not found: 'switch.seeedmultichannelrelay'.
  platform: seeedmultichannelrelay
  id: relay_1
  name: relay1
  channel: relay_1
  seeedmultichannelrelay_id: SeedMultiChannelRelay_ID
```

KoenBreeman, 2023-04-15:
Hi Cocoffr

Somehow it didn't sync my update yesterday. I've now performed the update again and tested it on my ESPHome setup. I removed the folders and it downloaded the files again. Also it builds.

INFO Reading configuration /config/test.yaml...
INFO Cloning https://github.com/esphome/esphome.git@pull/4684/head
INFO Fetching pull/4684/head
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: m5stack-atom; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 2.0.0
Compiling .pioenvs/test/src/main.cpp.o
Linking .pioenvs/test/firmware.elf
RAM:   [=         ]  12.5% (used 40936 bytes from 327680 bytes)
Flash: [=====     ]  50.6% (used 929009 bytes from 1835008 bytes)
Building .pioenvs/test/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/test/firmware.bin"], [".pioenvs/test/firmware.elf"])
Wrote 0xf4380 bytes to file /config/.esphome/build/test/.pioenvs/test/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 51.35 seconds =========================

********************************************************************************

Please let me know if it works for you as well.

cocoffr, 2023-04-15:
HI,
relay1 and relay2 are OK, when I drive the relay3, the relay4 moves, and when I drive relay4 nothing is moving

KoenBreeman, 2023-04-15:
Hi Cocoffr

I've updated the code. I found the bit description in the code from Seeed, but apparantly this wasn't correct. Could you check now if it's ok?

cocoffr, 2023-04-15:
wonderful, it's wok very well
thx very much
(excuse me for my english :)

KoenBreeman, 2023-04-16:
I've asked @jesserockz to review the PR and hopefully it will be merged into the new version.