esphome/feature-requests #2184: M5Stack 4 relays request

KoenBreeman, 2023-03-27:
**Describe the problem you have/What new integration you would like**
I have recently bought the M5Stack 4 relays unit. I have already build some custom code for myself and that works. However I thought that others might benefit from this as well. So perhaps this code (with improvements from others of course) can be implemented into ESPHome. 

This is the unit itself: 
https://docs.m5stack.com/en/unit/4relay

This is my github: 
https://github.com/KoenBreeman/ESPHome

**Please describe your use case for this integration and alternatives you've tried:**

It’s sets up 4 switches which can be used. Perhaps my code can better be set up. Perhaps it’s better to use a hub and set up the switch to have a channel. But I leave that up to the developers. 

**Additional context**

nielsnl68, 2023-03-27:
I have already an `external_component:` 
check out: https://github.com/nielsnl68/esphome-components/tree/main/components/m5stack4relay

KoenBreeman, 2023-03-27:
Hi Niels

I know that you have made this component. I used your code and made some modifications. But this request is to implement it natively into ESPHome so others can benefit from it as well without having to use custom components.

pricard32, 2024-12-20:
> Hi Niels
> 
> I know that you have made this component. I used your code and made some modifications. But this request is to implement it natively into ESPHome so others can benefit from it as well without having to use custom components.

Hi, How can I use your component?
The one from nielsnl68 do not work for me and I would like to test yours.
Thanks,

randybb, 2024-12-20:
@pricard32 everything you need is mentioned here https://github.com/esphome/esphome/pull/7234

pricard32, 2024-12-20:
Thanks a lot mate!

[Image]<https://assets.ad-da.ca/adda.jpeg>

Paul Ricard

Sans frais. :   1.833.234.ADDA
Cellulaire. :   1.819.995.5251
Courriel :   ***@***.*** ***@***.***>
Web :   ad-da.ca <https://ad-da.ca/>

closed 2026-10-22 by bdraco