esphome/feature-requests #2185: New internal ESPHome Trend Sensor requested

jazzmonger, 2023-03-28:
**Describe the problem you have/What new integration you would like**

Create a new "Trend sensor" in ESPHome based on an internal sensor value.

**Please describe your use case for this integration and alternatives you've tried:**

Creating an ESPHome template sensor based on a home assistant trend sensor works great as long as home assistant is available. 
**Additional context**

Only works if home assistant is available. My use case must work without home assistant being available,

basically implement the same functionality as exists in home assistant:
```
binary_sensor:
  - platform: trend
    sensors:
      house_stove_pipe_temp_falling:
        entity_id: sensor.house_uart_pipe_temp
        sample_duration: 10
        max_samples: 10
        min_gradient: -0.2 #5 degrees / 10s
        device_class: cold
    
      house_stove_pipe_temp_rising:
        entity_id: sensor.house_uart_pipe_temp
        sample_duration: 10
        max_samples: 10
        min_gradient: 0.2 #5 degrees / 10s
        device_class: heat 
```

Tuckie, 2023-04-23:
Yes please!  Trying to do a similar thing with a temperature sensor for a fireplace to verify state (aka fireplace lit).

Unfocused, 2023-04-23:
Have been thinking the same too, in my case for a smart climate control of a heat pump (rather than just a connected heat pump). 

A binary sensor would be a good basic case, but I'd also like a number sensor to report the estimated gradient. Of course, calculating that is pretty simple... it's dealing with a bunch of noisy measurements over time that's the hard part.

#667 seems like either a duplicate or at least related.