esphome/feature-requests #2186: Use Kelvin as the preferred color temperature unit

sebastianrasor, 2023-03-30:
**Describe the problem you have/What new integration you would like**
When communicating with Home Assistant through the `api` component, communication should be done in kelvin rather than mireds.

**Please describe your use case for this integration and alternatives you've tried:**
The maximum temperature of my device is 6500 kelvin, but I can only set it as high as 6493 kelvin due to unit conversion.
[image]

The same issue occurs for the lower bounds. The minimum for this device is 2700, however I can only set it as low as 2702 kelvin.
[image]

In addition, there's this frustrating visual effect when setting the temperature as a result of unit conversion.
[Screencast from 03-20-2023 08:24:30 PM.webm](https://user-images.githubusercontent.com/92653912/226520472-c3c00698-6bfe-4e76-8acc-7937f6941631.webm)

There are no alternatives to be tried other than simply not using ESPHome since mireds are the only unit used in communication with Home Assistant.

**Additional context**

sebastianrasor, 2023-03-30:
When you click a point on the color temperature gradient, the value initially changes to the point you clicked and then is quickly changed to a nearby value due to loss of precision through unit conversion.

danielkent-net, 2023-04-12:
I'm currently in the process of building a custom controller for a CWWW lamp, and while I can't test it right now (still waiting on parts), I might also be affected by this issue. My lamp only has a range of 3000-6000K, so the effects of floating point conversion might not be _quite_ as pronounced as yours, but I'm still concerned enough that I'd be willing to help out with this issue.

Home Assistant itself [moved away from mireds a while ago](https://github.com/home-assistant/architecture/discussions/564), so it might be a good idea to maintain unit parity with HA when possible. My main concern is that changing the units might break some existing configurations in non-obvious ways if they, for instance, set color temperature manually in lambda functions. An example might be an integration that increases color temperature by taking the existing color temperature and adding a value; if the units are changed to Kelvin, this function would actually _decrease_ color temperature (since adding to mireds is equivalent to reducing Kelvin color temperature), and much more slowly if using a fixed step value.

Based on what I've seen so far with the ESPHome code, it does at least support Kelvin as a valid color temperature unit in the YAML configuration, and correctly converts these values to mireds (1000000/K). However all of the internal functions still depend on units being in mireds. Some of the code looks like it will work correctly even if there's a unit change, but I'm not sure about the parts that use gamma correction.

I'll take a look at this again once I have enough parts where I can start testing on real hardware.

sebastianrasor, 2023-04-13:
I think the loss of precision could be solved by simply transmitting the mired value to Home Assistant as a float or double instead of an int. It would be nice to just up and switch to Kelvin across the board, but you're right about backwards compatibility for internal functions. Maybe just do what Home Assistant did and introduce *_kelvin variants of any color temperature related functions

danielkent-net, 2023-04-13:
> I think the loss of precision could be solved by simply transmitting the mired value to Home Assistant as a float or double instead of an int.

I looked into this possibility, and found that unlike the HA REST API, the [ESPHome protocol buffer API](https://github.com/esphome/esphome/blob/dev/esphome/components/api/api.proto#L459) in fact transmits color temperature as a float, but the value is [rounded to the nearest integer on the HA side.](https://github.com/home-assistant/core/blob/dev/homeassistant/components/esphome/light.py#L341) Changes might need to be made on both sides to support higher precision.

It also looks like this issue goes beyond just ESPHome devices, but rather to all devices that use mireds as their color temperature under the hood. When looking into this issue, I tried to set my Philips Hue color temperature to a specific value via the REST API (which does not accept float values; if you try, a 400 error results), and found that it rounded to the nearest whole temperature value in mireds:

```
~/.local/bin/hass-cli --server REDACTED -o yaml -x service call light.turn_on --arguments entity_id=light.bedroom_wall_light,color_temp_kelvin="4500"
   -  entity_id: light.bedroom_wall_light
      state: on
      attributes:
          min_color_temp_kelvin: 2202
          max_color_temp_kelvin: 6535
          min_mireds: 153
          max_mireds: 454
          supported_color_modes:
             -  color_temp
          color_mode: color_temp
          brightness: 254
          color_temp_kelvin: 4504
          color_temp: 222

```

bdraco, 2023-04-13:
> I looked into this possibility, and found that unlike the HA REST API, the [ESPHome protocol buffer [...]

I'd probably change the HA side to convert to kelvin first since its a float in esphome. If you still have precision loss than you could change the esphome side

nagyrobi, 2023-04-13:
I think an issue should be filled then...?

bdraco, 2023-04-13:
> I think an issue should be filled then...?
> 
> 

Technically the conversion loss is a bug so an issue makes sense on the face.

Unless someone plans on fixing it on the Hass side I wouldn't bother with an issue as it's likely a candidate for stale bot given it's pretty far down on the list and there are other issues in the queue that are causing breakage that will likely be prioritized higher.

nagyrobi, 2023-04-13:
Yeah that's what I was afraid of. This is a big project. @danielkent-net perhaps you could suggest a PR with a fix...?

danielkent-net, 2023-04-13:
I can take the lead on this issue. I'll submit an issue on the HA repo with an outline of the proposed fix, then do some experimenting to see if I can get it working the way it should before I start on a PR.

It might not be too big of a fix - I think [components/esphome/light.py:215](https://github.com/home-assistant/core/blob/dev/homeassistant/components/esphome/light.py#L215) just needs to be changed to check for `ATTR_COLOR_TEMP_KELVIN` by default, and only fall back to `ATTR_COLOR_TEMP` if the property doesn't exist.

danielkent-net, 2023-04-18:
It looks like adding Kelvin properties to the HA-side esphome light class instances solves most of the issues. There is still a little bit of visible value shifting on change, but it's only by one, and it's a result of using `floor()` instead of `round()` on converted fractional mired values (which is a potentially breaking PR that's out of scope of what's already been done). So, there's nothing that _needs_ to be done on the ESPHome side to support setting color temperatures without loss of precision.

With that said, it might still be useful to add helper functions within ESPHome to allow manipulating the color temperature value directly on the device in Kelvin instead of mireds. At the very least, it might be useful to have something like `{get,set}_color_temperature_kelvin()` to use in lambda functions. I would personally like to see a color temperature service equivalent to `light.dim_relative` (or even just add a `color_temperature` parameter to that function) so that manipulating color temperatures don't require as many lambda function calls.