esphome/feature-requests #2187: SIM800L send/receive SMS not only in English (PDU mode)

nordeep, 2023-03-31:
**Describe the problem you have/What new integration you would like**

Ability to send and receive SMS not only in English. SIM800L may send and receive in PDU mode using UCS2 format.
`AT+CMGF=0` - Turn on PDU mode.

**Please describe your use case for this integration and alternatives you've tried:**

Fully described with Arduino code available here - http://codius.ru/articles/GSM_%D0%BC%D0%BE%D0%B4%D1%83%D0%BB%D1%8C_SIM800L_%D1%87%D0%B0%D1%81%D1%82%D1%8C_3 (Russian language)

techmike78, 2024-01-08:
Hi All.
Please add support for PDU mode to this integration, for decoding just such a simple function is enough, you can put it in the HA itself, but it’s better to add it to the integration, where it is needed.

`String UCS2ToString(String s) {                       // Функция декодирования UCS2 строки
  String result = "";
  unsigned char c[5] = "";                            // Массив для хранения результата
  for (int i = 0; i < s.length() - 3; i += 4) {       // Перебираем по 4 символа кодировки
    unsigned long code = (((unsigned int)HexSymbolToChar(s[i])) << 12) +    // Получаем UNICODE-код символа из HEX представления
                         (((unsigned int)HexSymbolToChar(s[i + 1])) << 8) +
                         (((unsigned int)HexSymbolToChar(s[i + 2])) << 4) +
                         ((unsigned int)HexSymbolToChar(s[i + 3]));
    if (code <= 0x7F) {                               // Теперь в соответствии с количеством байт формируем символ
      c[0] = (char)code;                              
      c[1] = 0;                                       // Не забываем про завершающий ноль
    } else if (code <= 0x7FF) {
      c[0] = (char)(0xC0 | (code >> 6));
      c[1] = (char)(0x80 | (code & 0x3F));
      c[2] = 0;
    } else if (code <= 0xFFFF) {
      c[0] = (char)(0xE0 | (code >> 12));
      c[1] = (char)(0x80 | ((code >> 6) & 0x3F));
      c[2] = (char)(0x80 | (code & 0x3F));
      c[3] = 0;
    } else if (code <= 0x1FFFFF) {
      c[0] = (char)(0xE0 | (code >> 18));
      c[1] = (char)(0xE0 | ((code >> 12) & 0x3F));
      c[2] = (char)(0x80 | ((code >> 6) & 0x3F));
      c[3] = (char)(0x80 | (code & 0x3F));
      c[4] = 0;
    }
    result += String((char*)c);                       // Добавляем полученный символ к результату
  }
  return (result);
}`

l3zha, 2024-03-07:
> Found a temporary solution:
> 
> 1. created a new file next to the yaml project /homeassistant/esphome/sim800l_pdu.h
>    I placed in it only the code of the UCS2ToString decoding function and all its dependencies
> 
> ```
> /*
> Библиотека работы с SIM800 + arduino nano v3
> создана на основе примеров из статей цикла:
> http://codius.ru/articles/GSM_%D0%BC%D0%BE%D0%B4%D1%83%D0%BB%D1%8C_SIM800L_%D1%87%D0%B0%D1%81%D1%82%D1%8C_2
> 
> */
> 
> unsigned char HexSymbolToChar(char c) {
>   if      ((c >= 0x30) && (c <= 0x39)) return (c - 0x30);
>   else if ((c >= 'A') && (c <= 'F'))   return (c - 'A' + 10);
>   else                                 return (0);
> }
> 
> String UCS2ToString(String s) {                       // Функция декодирования UCS2 строки
>   String result = "";
>   unsigned char c[5] = "";                            // Массив для хранения результата
>   for (int i = 0; i < s.length() - 3; i += 4) {       // Перебираем по 4 символа кодировки
>     unsigned long code = (((unsigned int)HexSymbolToChar(s[i])) << 12) +    // Получаем UNICODE-код символа из HEX представления
>                          (((unsigned int)HexSymbolToChar(s[i + 1])) << 8) +
>                          (((unsigned int)HexSymbolToChar(s[i + 2])) << 4) +
>                          ((unsigned int)HexSymbolToChar(s[i + 3]));
>     if (code <= 0x7F) {                               // Теперь в соответствии с количеством байт формируем символ
>       c[0] = (char)code;                              
>       c[1] = 0;                                       // Не забываем про завершающий ноль
>     } else if (code <= 0x7FF) {
>       c[0] = (char)(0xC0 | (code >> 6));
>       c[1] = (char)(0x80 | (code & 0x3F));
>       c[2] = 0;
>     } else if (code <= 0xFFFF) {
>       c[0] = (char)(0xE0 | (code >> 12));
>       c[1] = (char)(0x80 | ((code >> 6) & 0x3F));
>       c[2] = (char)(0x80 | (code & 0x3F));
>       c[3] = 0;
>     } else if (code <= 0x1FFFFF) {
>       c[0] = (char)(0xE0 | (code >> 18));
>       c[1] = (char)(0xE0 | ((code >> 12) & 0x3F));
>       c[2] = (char)(0x80 | ((code >> 6) & 0x3F));
>       c[3] = (char)(0x80 | (code & 0x3F));
>       c[4] = 0;
>     }
>     result += String((char*)c);                       // Добавляем полученный символ к результату
>   }
>   ESP_LOGI("main", "len result %1d",result.length());
>   ESP_LOGI("main", "len s %1d",s.length());
>   
>   if (result.length()>0) {
>    ESP_LOGI("main", "text is ucs2");
>    return (result);
>   }else {
>    ESP_LOGI("main", "text is not ucs2");
>    return (s);}
> }
> ```
> 
> 2. in the yaml of the project I connected sim800l_pdu.h in this way:
>    esphome:
>    includes:
>    
>    * sim800l_pdu.h
> 3. in the project method I used the conversion function itself like this (type conversions are important):
>    on_ussd_received:
>    
>    * logger.log:
>      format: "ussd is %s"
>      args: [ 'UCS2ToString(ussd.c_str()).c_str()']
>    * lambda: |-
>      id(ussd_message).publish_state(UCS2ToString(ussd.c_str()).c_str());

Можете мне помочь? Правильно ли я использовал ваш способ? Это автоматизация должна пересылать входящие смс на мой номер, но вместо "Привет" получаю "@@Ñ@@@@@@@Ñ@"

```
sim800l:
  on_sms_received:
    then:
    - sim800l.send_sms:
        recipient: '+7**********'
        message: !lambda 'return (UCS2ToString(message.c_str()).c_str());'
```

latonita, 2024-03-17:
>     - sim800l.send_sms:
>         recipient: '+7**********'
>         message: !lambda 'return (UCS2ToString(message.c_str()).c_str());'
> ```

Have you tried just forward a message, without using UCS2ToString?
--
Вы не пробовали просто пересылать сообщение, без применения функции UCS2ToString ?

l3zha, 2024-03-18:
> > ```
> > - sim800l.send_sms:
> >     recipient: '+7**********'
> >     message: !lambda 'return (UCS2ToString(message.c_str()).c_str());'
> > ```
> 
> ## Have you tried just forward a message, without using UCS2ToString?
> Вы не пробовали просто пересылать сообщение, без применения функции UCS2ToString ?

```
    - sim800l.send_sms:
        recipient: '+7*******'
        message: !lambda 'return message;'
```

If so, then instead of "Привет" I get "041F04400438043204350442"

latonita, 2024-03-18:
looks like component always sends sms in text mode. 
I found an article in russian, second half explains how to send in PDU/UCS2 format. http://codius.ru/articles/GSM_%D0%BC%D0%BE%D0%B4%D1%83%D0%BB%D1%8C_SIM800L_%D1%87%D0%B0%D1%81%D1%82%D1%8C_3
I checked my sim800l - and its damaged, so i can't test it

I think i found a code from the article on the github for reuse
https://github.com/evgenysx/cek-iot/blob/8ed2d83c489560cb3f78c71e232905b6a5ec9fa5/lib/sim800l/gprs.cpp

behnam85, 2024-09-01:
I have the same problem. Please add PDU mode.🙏