esphome/feature-requests #2190: Whether the web page of esphome can be embedded to play the camera stream. Similar to this GitHub - AlexxIT/WebRTC: Home Assistant custom component for viewing almost any camera stream in real time using WebRTC and other technologies

cmsjin, 2023-03-31:
**Describe the problem you have/What new integration you would like**

Whether the web page of esphome can be embedded to play the camera stream. Similar to this [GitHub - AlexxIT/WebRTC: Home Assistant custom component for viewing almost any camera stream in real time using WebRTC and other technologies](https://github.com/AlexxIT/WebRTC)
**Please describe your use case for this integration and alternatives you've tried:**

I hope the web page can embed the camera stream in this way
https://community-assets.home-assistant.io/original/4X/0/7/9/0793909dd1b93acf6a1c65bd170a7bd39e96581a.jpeg
**Additional context**

nagyrobi, 2023-03-31:
See: https://esphome.io/components/esp32_camera_web_server.html
The webserver of ESPHome is not designed to draw fancy UIs with pictures. These MCUs have little resources for such things.

cmsjin, 2023-04-06:
> 参见： https://esphome.io/components/esp32_camera_web_server.html ESPHome 的网络服务器不是为绘制带有图片的花哨 UI 而设计的。这些MCU几乎没有资源来处理这些事情。

Perhaps my expression was incomplete or you misunderstood. I don't need Ephome to provide a video stream, just play the stream.