esphome/feature-requests #2191: Add support for 'WeAct Studio Epaper Module' screen

lboue, 2023-03-31:
**Describe the problem you have/What new integration you would like**

[WeAct Studio Epaper](https://github.com/WeActStudio/WeActStudio.EpaperModule) screen is not recognized well in ESPHome and it generates errors:

- Display color:  black, white
- Size: 122x250
- Full refresh time: 8s
- Driver chip：SSD1680
- Display Driver：DEPG0213BN

**Please describe your use case for this integration and alternatives you've tried:**

- Home Assistant 2023.3.6
- ESPHome version: 2023.3.2

The screen is connected to an ESP-C3 board
```
Setup SPI
spi:
  clk_pin: 4
  mosi_pin: 6

# Configuration for the display
display:
  - platform: waveshare_epaper
    model: 2.90inV2
    cs_pin: 7
    dc_pin: 1
    busy_pin: 3
    reset_pin: 2
    rotation: 270°
    id: my_display
    lambda: |-
      it.printf(125, 6, id(font1), TextAlign::TOP_CENTER, "%s", "Hello World!");
      it.strftime(125, 116, id(font1), TextAlign::BOTTOM_CENTER, "%Y-%m-%d %H:%M", id(time_homeassistant).now());
```

**Additional context**

This screen works fine with this exemple '[EpaperModuleTest_Arduino_ESP32C3.ino](https://raw.githubusercontent.com/WeActStudio/WeActStudio.EpaperModule/master/Example/EpaperModuleTest_Arduino_ESP32C3/EpaperModuleTest_Arduino_ESP32C3.ino)' with Arduino IDE as a demo.

Unfocused, 2023-04-23:
I just stumbled across these too and wondered how well it'd be supported by ESPHome, since they're quite well priced. They actually have four ePaper modules:

- 2.13inch, 250px x 122px, black/white 
- 2.13inch, 250px x 122px, black/white/red
- 2.9inch, 296px x 128px, black/white
- 2.9inch, 296px x 128px, black/white/red

Product page: https://www.aliexpress.com/item/1005004644515880.html

All are driven by the same [SSD1680](https://github.com/WeActStudio/WeActStudio.EpaperModule/blob/master/Doc/SSD1680.pdf), which supports both 4-wire SPI and 3-wire SPI. 

Based on the ESPHome documentation alone, I wouldn't have known to check the `waveshare_epaper` display platform - it doesn't mention that it supports SSD1680 or DEPG0213BN. In fact, SSD1680 is only in the code as part of a URL to a datasheet in a comment for the Good Display 2.9in black/white/grey. And I know the LilyGo T5 v2.3.1 uses the same driver & display module. It'd be helpful if the docs listed the drivers & display modules known to work, as well as the products. I know there can still be subtle differences & incompatibilities, but at least it would be a useful pointer for what to try.

There's actually a report that the WeAct 2.9inch black/white model is working already:
https://community.home-assistant.io/t/weact-2-9-epaper-module-working/539413

@lboue - your comment says you have the 2.13inch model, but your example code is setup for a 2.9inch model, which has a different resolution. If that's the case, then that's likely your issue. I'd suggest trying the various 2.13inch models instead (`2.13in`, `2.13in-ttgo`, etc) to see if one of them works. If that works, we could get the documentation updated. And maybe I'll order one myself.

lboue, 2023-04-23:
I tried with parameter set to 2.13inch but it doesn't work a all.

Unfocused, 2023-04-27:
> I tried with parameter set to 2.13inch but it doesn't work a all.

To be clear: Did you try all the different 2.13in options? There's 6 different models for that screen size: https://esphome.io/components/display/waveshare_epaper.html?highlight=waveshare#configuration-variables

lboue, 2023-04-27:
yes I tested the different choices and the one that works best is 2.90inV2.

lboue, 2023-04-28:
| model  | screen size | screen printing |
| --------------- | --------------- | --------------- |
| 2.13in-ttgo | KO | OK |
| 2.90inV2 | KO | KO |

lboue, 2023-04-28:
> There's actually a report that the WeAct 2.9inch black/white model is working already: https://community.home-assistant.io/t/weact-2-9-epaper-module-working/539413

@Unfocused How can a add a new model profile similar to 2in9_V2 but for 2.13inch?

> @lboue - your comment says you have the 2.13inch model, but your example code is setup for a 2.9inch model, which has a different resolution. If that's the case, then that's likely your issue. I'd [...]
It doesn't work with 2.13in profiles.

Unfocused, 2023-05-06:
I wish there was a simple answer to that... 

But it is easier than starting from scratch.

A new model would need to be added to the list of models accepted in the YAML file here:
https://github.com/esphome/esphome/blob/3498aade85f10e342efcd03bccc1dcdeceff5b5f/esphome/components/waveshare_epaper/display.py#LL70C0-L94C0

It seems all those similar displays are implmemented by the `WaveshareEPaperTypeA` class. So the new model needs adding to the enum here:
https://github.com/esphome/esphome/blob/3498aade85f10e342efcd03bccc1dcdeceff5b5f/esphome/components/waveshare_epaper/waveshare_epaper.h#LL73C0-LL83C0

And the `WaveshareEPaperTypeA` class needs to handle that models differently somehow:
https://github.com/esphome/esphome/blob/3498aade85f10e342efcd03bccc1dcdeceff5b5f/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L164

The hard part is figuring out the correct LUT for that display, and how the commands are different from the other models. As you can see here, the LUTs are just a bunch of numbers:
https://github.com/esphome/esphome/blob/3498aade85f10e342efcd03bccc1dcdeceff5b5f/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L164

But the manufactuers often keep these secret. And when they do publish code, it's often a mess. I haven't checked, but hopefully everything needed is in [WeAct Studio's GitHub repository](https://github.com/WeActStudio/WeActStudio.EpaperModule) - you'd need to figure out whats the same and what's different from the other models that the `waveshare_epaper` component supports. Hopefully it's just the startup sequence that's different.

lboue, 2023-05-07:
Here is the result I get with my 2.13inch, 250px x 122px, black/white module:
[image]

With this config:

```
# 22 BUSY	Busy status output
# 19 RST	External reset, low active
# 23 DC	Data / Command selection (high for data, low for command)
# 33 CS	SPI chip selection, low active
# 21 SCK	SPI SCK pin
# 25 DIN	SPI MOSI pin
#   GND	Ground
#   VCC	3.3V/5V

# Pins for Waveshare ePaper ESP Board
spi:
  clk_pin: GPIO21
  mosi_pin: GPIO25
  
display:
  - platform: waveshare_epaper
    id: eink_display
    busy_pin: GPIO22
    reset_pin: GPIO19
    dc_pin: GPIO23
    cs_pin: GPIO33
    model: 2.13in-ttgo-dke
    update_interval: 60s
    rotation: 270°
    lambda: |-
      it.print(0, 0, id(font1), "ESPHome ePaper screen");
      it.printf(0, 10, id(font1), "ip: %s",id(my_ip).state.c_str());
      it.printf(0, 25, id(font1), "rssi: %.2f",id(inside_temperature) );
      it.strftime(0, 40, id(font1), "%Y-%m-%d %H:%M", id(esptime).now());
```

Unfocused, 2023-05-10:
Oh! That's interesting! 

Unlike other displays named like it, the `2.13in-ttgo-dke` model is implemented in the `WaveshareEPaper2P13InDKE` class - it's different from all other models (I assume it's a newer version). And `2.13in-ttgo-dke` model has a resolution of 128x250, whereas the resolution of the WeAct screen is 122x250. Coincidentally, the strip of corruption at the top of your screen looks like it might be 6 pixels high (128 - 122 = 6). So hopefully it'd require a simpler change to `WaveshareEPaper2P13InDKE`, possinly even just changing t eh resolution it uses for that model.

Unfocused, 2023-05-10:
I just saw there's an existing bug for support for the WeAct Studio ePaper displays (remember to search for existing issues first!): 
https://github.com/esphome/feature-requests/issues/2101

In the other issue, they seem to come to the same conclusion.

In fact, there's even an existing pull request that adds support for these displays:
https://github.com/esphome/esphome/pull/4579/
(Although it's more complicated than just changing the resolution)

Try it out using:
```yaml
external_components:
  - source: github://pr#4579
    components: [waveshare_epaper]

display:
  - platform: waveshare_epaper
    model: 2.13inbn
    id: eink_display
    busy_pin: GPIO22
    reset_pin: GPIO19
    dc_pin: GPIO23
    cs_pin: GPIO33
    update_interval: 60s
    lambda: !lambda |-
      it.print(0, 0, id(font1), "ESPHome ePaper screen");
```

And report your findings in #2101

thijsk, 2023-09-12:
I just tried that PR with a LOLIN SSD1680 based 2.13" display, and it does not work for me. 
Display does not update and I get warnings in the log:
```
[14:43:49][I][waveshare_epaper:1881]: Performing partial e-paper update.
[14:43:49][I][waveshare_epaper:1989]: Completed e-paper update.
[14:43:49][W][component:204]: Component waveshare_epaper.display took a long time for an operation (0.65 s).
[14:43:49][W][component:205]: Components should block for at most 20-30ms.
```

```
display:
  - platform: waveshare_epaper
    model: 2.13inbn
    id: eink_display
    cs_pin: D0
    dc_pin: D8
    update_interval: 60s
    lambda: |-
      it.printf(0, 0, id(roboto), "Hello World!");
      ```
      
    I did get the display working with the arduino example code from https://github.com/wemos/LOLIN_EPD_Library, with the same pinout on the same esp device.

kungjohan, 2024-01-12:
After updating to ESPHome 2023.12.0 I get the error ValueError: Component ID waveshare_epaper_waveshareepaper was not declared to inherit from Component, or was registered twice.

As described in:
https://github.com/esphome/issues/issues/5256

"There was a change in the display structure that breaks all external display components; the fix needs to be done in the external component. See https://github.com/landonr/lilygo-tdisplays3-esphome/pull/46 for an example of the needed changes to have this compile again"

Is it possible to update this pull request with the needed changes? Now I cannot use my display after updating ESPhome

t4ndu, 2024-02-12:
Can i use 8x 2.9"EINK Model over SPI, can this Weact E INK Display handle daisy chain? i want create an ticker display like the Blockclock with 8 digits.

jbergler, 2024-02-14:
If anyone wants to try the 2.9" 3-color WeAct display, would love some testing on my PR.

```yaml
external_components:
 - source: github://pr#6626
   components: [waveshare_epaper]

display:
  - platform: waveshare_epaper
    model: 2.90in3c
    # the rest of your display config as normal
```

PsY4, 2024-02-19:
Works great ! Thx!

regystro, 2024-03-19:
I also can confirm it works flawlessly:

[image]

Beware of the typo in the sample: it's `pr#6226` instead of `source: github://pr#6626`

Thank you for you effort and for sharing!!

victorclaessen, 2024-03-20:
I can confirm that this now also successfully runs color on the **2.13"** version. However, perhaps because I use model `2.90in3c` from `pr#6226`, I have to skip the first line (i.e. start at _y_ value of **1**) when using it at 90 degrees. (You can see the difference with the two letters **E** in the top right corner.)

[image]

```
external_components:
  - source: github://pr#6226
    components: [waveshare_epaper]
font:
  - file: 'arial.ttf'
    id: arial
    size: 24    
color:
  - id: red
    hex: "FF0000"
  - id: black
    hex: "FFFFFF"
display:    
  - platform: waveshare_epaper
    model: "2.90in3c"
    rotation: 90
    lambda: !lambda |-
      it.print(0, 4, id(arial), id(red), "ESPHome ePaper");
      it.print(220, 1, id(arial), id(red), "E");
      it.print(235, 0, id(arial), id(black), "E");
      it.printf(10, 60, id(arial), "uptime: %s",id(uptime_text).state.c_str());
```

nagyrobi, 2024-03-21:
Wow, great!
I just ordered a 2.9" 3-color one for testing with [LVGL](https://github.com/esphome/esphome/pull/6363).

closed 2024-04-22 by lboue