esphome/feature-requests #2194: Compile against custom fork of esp-idf

tracestep, 2023-04-03:
**Describe the problem you have/What new integration you would like**

I've hit a bug in the `esp-idf` library (which is also manifests itself with the Arduino framework, by the way). There are a few patches available, but I don't know if it would be possible to specify a custom fork of esp-idf for ESPHome to use instead of the official one. If not, I think it would be nice to implement, much like `external-components` work.

**Please describe your use case for this integration and alternatives you've tried:**

`i2c_master_cmd_begin(port_, cmd, 20 / portTICK_PERIOD_MS)` call in `i2c_bus_esp_idf.cpp` from the `i2c` component blocks for a full second if the device does not respond, which wreaks havoc with other tasks and components used.
See [https://github.com/espressif/esp-idf/issues/4999](https://github.com/espressif/esp-idf/issues/4999)

**Additional context**

The issue is fixed / worked around on a few patches, but still not fixed on `esp-idf` itself.

tracestep, 2023-04-03:
I am actually using the esp-idf framwork itself on ESPHome, I just tested with the Arduino to see if the issue manifested itself there as well.

ssieb, 2023-04-03:
Then you can specify whatever version of esp-idf you want to use:
https://esphome.io/components/esp32.html#esp-idf-framework
See the `source:` option.  I'm not sure of the exact syntax for that though.

tracestep, 2023-04-03:
Oops, guess I missed that option (I must have glanced only on the example). Thanks! Closing the issue.

closed 2023-04-03 by tracestep