esphome/feature-requests #2196: EspHome enabled wifi range extender, but missing nat

Samu84, 2023-04-06:
**Describe the problem you have/What new integration you would like**
It would be nice to enable a range extender using and esp8266 and esp8285, by pressing one GPIO, i am able to enable the function, but without NAT option, there are not data, is only connecting, i found this way to enable range extender studing arduino code.
On platformio.io there is the library and there is on github also the file used to enable it on arduino:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip2/include/lwip/napt.h 
Tasmota and arduino are able to enable an esp as wifi range extender, it would be nice if also esphome is able to do it, i setup a lot of esp  with esphome, is more simple to program and you can do everything you need. that's why i decided to move from arduino ide to esphome, arduino was taking a lot of time to make all the code, esphome is much faster.

**Please describe your use case for this integration and alternatives you've tried:**
binary_sensor:
  - platform: gpio
    name: WiFi Extender
    internal: false
    pin:
      number: GPIO01
      inverted: true
      mode:
        input: true
        pullup: true
    on_press:
      then:
        lambda: |-
         WiFi.setPhyMode(WIFI_PHY_MODE_11N); enable wifi to N
         WiFi.mode(WIFI_AP_STA); Enable AP and STA
         WiFi.softAPConfig(
           IPAddress(192, 168, 4, 1),
           IPAddress(192, 168, 4, 1),
           IPAddress(255, 255, 255, 0)); Config IP, Gateway and Subnet
         WiFi.softAP("APtest"); Give a name to AP (optional also password)
    on_release:
      then:
        lambda: |-
         WiFi.mode(WIFI_STA); Disable AP

**Additional context**

nagyrobi, 2023-04-06:
Using "range extenders" is usually a cheap and bad practice, because they being all independent from the "master" AP, they increase noise and interference and in the end, you get with worse results than you had before, without any extension.
Not to speak about the performance loss caused by an ESP MCU thin data throughtput.

Samu84, 2023-04-06:
i think is nice to have it when you need to extend the range between more esp in a big house, avoiding to buy Wifi range extender, if is about internet navigation, yes that's right is not so good, is only about 5Mbit speed, at the moment i am using 2 with tasmota and 3 with arduino, this way i am able to arrive to almost 1km of distance ( no walls), sometimes some of them lose connection but is only about 5s 2 max 3 times day, same problems, sometimes worse, performance i would have with an AsusRange extender or also D-link, both when they start to lose signal can happens 20 times in 5minutes, and honestly i have more stability using tasmota or arduino

nagyrobi, 2023-04-06:
Asus/DLink etc such range extenders are just as weak, forget them.

Samu84, 2023-04-06:
yes but better performance i have with esp tasmota or arduino

nagyrobi, 2023-04-06:
You misunderstand. All domestic range extenders (including Asus/DLink/Tasmota/Arduino etc) make more bad than good on long term, increasing interference and making APs "fighting with each other". You shouldn't look on individual results, but overall solution. You'll end up in devices randomly connecting to one and another decreasing stability and reliability. And not only for you, but also for your neighbors.

Proper Wi-Fi coverage needs coordinated access points, where they are aware of each other etc.

Samu84, 2023-04-06:
Ok now i understand, but it would be nice this option when i need to test esphome to someone else and have temporary coverage where the wifi is missing

Regdr, 2023-04-06:
wow this is a subject i am interested too, at the moment, between the router I have an esp8285 then the esp32, but the esp32 is not receiving signal because is too far, so i needed to make it completely independent from HA, MQTT, etc.., I tried to program the esp8285 with tasmota range extender, with this option the esp32 is working, but i do not like it, because i need to make all the automations on HA, instead with esphome i can create all the automations and use HA just to monitor it, if the esp8285 would be able to operate like normally with the esphome plus the range extender would be perfect, so the esp32 can take the wifi signal and send the data to HA.

Jan3726, 2023-04-09:
> **Describe the problem you have/What new integration you would like** It would be nice to enable a range extender using and esp8266 and esp8285, by pressing one GPIO, i am able to enable the [...]

OMG. that's a nice idea i am looking from really a lot of time, I tried to use your way, but yes, there is the problem with NAT, but I think some files need to be added on esphome, I tried with the files that are needed for Arduino to work like RE, using includes: on esphome, but are giving a lot of errors
ESP8266WiFi.h - I think is included in ESPHOME
ESP8266WebServer.h - I think is included in ESPHOME
lwip/napt.h - needs to be included in ESPHOME
lwip/dns.h - needs to be included in ESPHOME
LwipDhcpServer.h - needs to be included in ESPHOME

mag1024, 2023-07-09:
@nagyrobi, you're right that many people have unrealistic expectations of what repeaters can accomplish, but the question is not whether they are the _best_ solution, but rather whether they are a useful capability. If my sprinkler controller is having wifi issues, I _could_ add an extra access point for the sake of this one device, or I could use a nearby esp32 to relay the minimal amount of bandwidth required from around the corner.

Anyway, I hacked up something that seems to kinda work: [mag1024/esphome-nat-ap](https://github.com/mag1024/esphome-nat-ap). _Caveat emptor._

Regdr, 2023-07-09:
Hello that's nice, at the moment i would want to try but the sp32 is in another home, now i am using an esp8285 (sonoff mini) because is included power supply and everything in a really small space, so i can hide behind a plug, this evening i try to include all the *.h required inside nat-ap.h

Many thanks for this help [mag1024](https://github.com/mag1024)

nagyrobi, 2023-07-09:
How do you forward the port back towards HA, if the client is also an ESPHome node?

mag1024, 2023-07-09:
> How do you forward the port back towards HA, if the client is also an ESPHome node?

I haven't tried it, but [ip_portmap_add()](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/lwip_napt.h#L100-L112) looks like it should do the trick.

nagyrobi, 2023-07-09:
With static lease + port forwading configurable, this could be a viable option. But bridge mode would be even better.

Jan3726, 2023-07-10:
OMG!!! finally some good news, I am using and esp8285 too (because of the same reasons, power suplply and small space), i tried every way to make it work but i got the same problem like [Regdr](https://github.com/Regdr), then i tried also to add the *.h files one by one, but some file is not possible to find in some repository

Jan3726, 2023-07-31:
Till now i tried every way to make this function work, but there is no way to exit, unfortunately i am using already 75 esp8285 and esp8266 and move from those to esp32, it would take a looot of time to solder new board and update HomeAssistant