esphome/feature-requests #2197: Captive Portal does not support Ethernet module

pgwipeout, 2023-04-06:
**Describe the problem you have/What new integration you would like**

Support the captive portal module with the ethernet module.

**Please describe your use case for this integration and alternatives you've tried:**

It is impossible to use the captive portal module with an ethernet based esphome device. Currently the captive-portal module is dependent on the wifi module, which cannot be loaded at the same time as the ethernet module.

**Additional context**

Meetsch, 2023-04-26:
What issue do you encounter with ethernet connection that you would like to solve with the captive portal?

I see 2 possibilitées:
1) change ethernet configuration (ip address)
2) OTA update
But the issue is that a "captive" ethernet access is tough... you would likely need to unplug the device trom you network as it will conflict with your router. Therefore I would recommend updating your device with USB to solve the ip configuration issue.

So it would be great if you try to explain your use case where you need a "captive portal" for ethernet.

ed-so, 2024-04-29:
I am taking up the old feature request. 
A captive portal or similar is required to connect an Ethernet only device  to the router. 
Currently only DHCP can be configured to connect to other networks. 
A change to fixed addresses is not possible for the user and therefore the device cannot be used in many networks.
The "# optional manual IP:"  is useless as it cannot be set by the users.

The portal should at least make IP,GW,SUBNET,DNS configurable.