esphome/feature-requests #2198: Home Assistant logbook entry action

clinta, 2023-04-07:
**Describe the problem you have/What new integration you would like**

I'd like an action for creating entries in the home assistant logbook associated with a specific entity. Something like

```yaml
- homeassistant.logbook_entry:
    id: my_sensor
    message: My sensor did something
```

This would fire a [`logbook_entry`](https://www.home-assistant.io/docs/configuration/events/#logbook_entry) event in home assistant with the entity_id properly populated.

**Please describe your use case for this integration and alternatives you've tried:**

I have some automatons in esphome, it would be valuable to have logs associated with these automations in home assistant. So for example I can look at the details of my light, and see which automation turned it on.

**Additional context**

I can almost get there firing a custom event, then setting up an automation in home assistant to listen for those events and create the logbook entries. The blocker is any way to reliably get the entity id.

```yaml
        - homeassistant.event:
            event: esphome.log_message
            data:
              entity_id: !lambda return id(my_sensor).get_object_id();
              message: here's a message
```

get_object_id() will sometimes work, but not if the entity id has been renamed in home assistant.

designerferro, 2024-04-27:
This would be vary helpfull on Bluetooth proxy from ESPhome. I have a button on a switchbot and would like to have an LOGBOOK_ENTRY each time the ESPhome BLE proxy connected to that switchbot to press the button.

rubin110, 2025-03-11:
My garage door randomly opened up last night at 2 AM. The ESP device hooked up to it is a simple relay with a hall sensor on the door. The hall sensor provides state to the cover entity. I wish there was some logging to know if the cover entity was changed due to some interaction through HA, or if the hall sensor detected the door being open and made the state change on its own.

Thankfully nothing was amiss in the garage outside a bunch of leaves blowing in.