esphome/feature-requests #2199: Remote Transmitter: NECX / Extended NEC protocol

iLeopod, 2023-04-08:
**Add NECX / Extended NEC protocol to remote Transmitter **

**Sending RAW ir codes works **

Extended nec has a 16 bit address and a 8 bit command. This is encoded as a 24 bit value as you would expect, with the lower 8 bits the command and the upper 16 bits the address.
https://www.sbprojects.net/knowledge/ir/nec.php

sasodoma, 2024-04-27:
Given that the current implementation of NEC protocol already uses 16 bit address and 16 bit command, with no limitations for the relation between higher and lower bits, I believe that Extended NEC can already be used normally.

ToBeHH, 2025-05-02:
Looking forward for NECext as well

DjordjeMandic, 2025-05-26:
Current nec implementation already supports extended protocol (16bit address).

esphome/esphome#8015 has bit more logging if you need to see it in logs.

You just need to supply your 16bits for address and 16bits for command in config (both as hex). There is no automatic padding for command or address so you have to calculate inverted byte by yourself.

Examples below are taken from description of esphome/esphome#8015

```yaml
# 16bit address (extended nec) example 
on_press:
  - remote_transmitter.transmit_nec
      address: 0x6B86
      command: 0xF30C
```

```yaml
# 8bit address (normal nec) example 
on_press:
  - remote_transmitter.transmit_nec:
      address: 0xBF40
      command: 0xF40B
```

Do you have any suggestions for different way to supply data in yaml config?