esphome/feature-requests #2200: Renogy BT-1 BLE support

rb2k, 2023-04-09:
Renogy has a series of [MPTT solar charge controllers](https://www.renogy.com/wanderer-10a-pwm-charge-controller/).
They have an RS232 port that one can hook up a bluetooth module to: the [BT-1](https://www.renogy.com/bt-1-bluetooth-module-new-version/)

Apparently they use BLE and there's a project that decodes it: https://github.com/cyrils/renogy-bt1
Request construction is over here: https://github.com/cyrils/renogy-bt1/blob/main/renogybt/Utils.py#L54

With the recent Homeassistant / BLE / Esphome advances, I'm not sure if something like that would belong in Homeassistant or in Esphome, but I thought I'd open a feature request just in case other people are looking for it.

jwmcintosh, 2023-07-05:
Seconding. This would be nice to have. I would like to be able to track various metrics such as solar production on a given day. The Renogy app is not great and requires my mobile device be nearby. Using a bluetooth proxy I could, in theory, place it near enough to the solar farm and pull data to log into Home Assistant (or other tool of choice).

mavenius, 2023-10-02:
Just putting it out there that the BT-2 module would be nice to see integrated as well :-)

I have 3 batteries daisy-chained (actively working with the cyrils/renogy-bt project to get them all auto-discovering) and it would be great to see them pulled into esphome.

riker65, 2023-11-19:
would be also interessted:-)

ajdezigns, 2023-12-17:
Yes please!

ajdezigns, 2023-12-24:
Hey everyone i found an awesome solution to our issues https://www.tindie.com/products/westlabs/renogy-rover-rs232-compatible-wifi-module/